prusa3d/PrusaSlicer #2557: PrusaSlicer does not open if the saved window position isn't on the main Monitor

GitSoks, 2019-06-25:
### Version
PrusaSlicer-2.0.0+win64-201905201652

### Operating system type + version
Windows 10 Professional 64 bit

### 3D printer brand / version + firmware version (if known)
Genuine Prusa i3 MK2.5, firmware 3.7.1

### Behavior
* **Problem**: Every time the PrusaSlicer window tries to open in any other Monitor then the main one (I have 3 monitors connected) it just gets stuck while loading the Interface (= a white Window opens without the Interface and the Mouse pointer indicates loading)

The only solution if this happens was for me to delete all PrusaSlicer config files in the "AppData" Windows folder so it opens on the main monitor again (resets the window Position do default = main monitor)

* **Workaround**:I have to always make sure I keep the PrusaSlicer window on the main Monitor and close it while it is on the main one. If I do that the next time I open PrusaSlicer, It has kept the Position and opens on the main Monitor again and everything works fine.

Braddg, 2019-06-25:
Duplicate of #2522

bubnikv, 2019-06-26:
Would you please post the PrusaSlicer.ini file that you need to delete to be able to restart the application?

What graphics cards are you running on? How many do you have in your computer? How many are active?

closed 2026-09-29 by prusaslicer-bot[bot]