prusa3d/PrusaSlicer #2558: Multicolor Extruder (Extruder 10-16 are not usable - therefore ever Extruder 1 (T0))

a-sprenger, 2019-06-25:
### Version
2.0.0+win64

### Operating system type + version
Win10 1809 64-bit

### 3D printer brand / version + firmware version (if known)
own built Prusa with repetier v1.0.4 

### Behavior
I have developed a multicolor-Extruder. When using different Colors (Extruders) the slicer uses the Extruder 1-9.
If I choose Extruder 10-16 the slicer slices instead of 10-16 ever Extruder 1 (T0 in gcode)

#### Project File (.3MF) where problem occurs

####Links to my water-cooled multicolor-Extruder####
https://www.thingiverse.com/thing:3590034

[Schiff_Schiff.zip](https://github.com/prusa3d/PrusaSlicer/files/3325555/Schiff_Schiff.zip)

lukasmatena, 2019-06-26:
@bubnikv There is something rotten. This is what ASAN thinks about OPENING than AMF (so I didn't even get to slice it):

=================================================================
==9156==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000088 (pc 0x000001bfe8c5 bp 0x7ffef4b1e6b0 sp 0x7ffef4b1e6b0 T0)
    #0 0x1bfe8c4 in stl_translate_relative(stl_file*, float, float, float) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x1bfe8c4)
    #1 0xc04f16 in Slic3r::TriangleMesh::translate(float, float, float) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0xc04f16)
    #2 0x98791e in Slic3r::ModelVolume::center_geometry_after_creation() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x98791e)
    #3 0x924d2f in Slic3r::AMFParserContext::endElement(char const*) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x924d2f)
    #4 0x18aa517 in doContent src/expat/expat/lib/xmlparse.c:2956
    #5 0x18aaf5b in contentProcessor src/expat/expat/lib/xmlparse.c:2533
    #6 0x18af3b3 in XML_ParseBuffer src/expat/expat/lib/xmlparse.c:1985
    #7 0x91f992 in Slic3r::load_amf_file(char const*, Slic3r::DynamicPrintConfig*, Slic3r::Model*) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x91f992)
    #8 0x920419 in Slic3r::load_amf(char const*, Slic3r::DynamicPrintConfig*, Slic3r::Model*) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x920419)
    #9 0x991203 in Slic3r::Model::read_from_file(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Slic3r::DynamicPrintConfig*, bool) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x991203)
    #10 0xfd1788 in Slic3r::GUI::Plater::priv::load_files(std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > const&, bool, bool) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0xfd1788)
    #11 0xfd61dd in Slic3r::GUI::Plater::load_files(std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > const&, bool, bool) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0xfd61dd)
    #12 0xfd6f71 in Slic3r::GUI::Plater::load_project() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0xfd6f71)
    #13 0x1877763 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) src/common/event.cpp:1396
    #14 0x1878010 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) src/common/event.cpp:1868
    #15 0x187810e in wxEvtHandler::TryHereOnly(wxEvent&) src/common/event.cpp:1589
    #16 0x18781a5 in wxEvtHandler::TryBeforeAndHere(wxEvent&) include/wx/event.h:3873
    #17 0x18781a5 in wxEvtHandler::ProcessEventLocally(wxEvent&) src/common/event.cpp:1526
    #18 0x1878220 in wxEvtHandler::ProcessEvent(wxEvent&) src/common/event.cpp:1499
    #19 0x18787a6 in wxEvtHandler::SafelyProcessEvent(wxEvent&) src/common/event.cpp:1617
    #20 0x168308b in wxMenuBase::DoProcessEvent(wxMenuBase*, wxEvent&, wxWindow*) src/common/menucmn.cpp:666
    #21 0x168317a in wxMenuBase::SendEvent(int, int) src/common/menucmn.cpp:643
    #22 0x160bac3 in menuitem_activate src/gtk/menu.cpp:543
    #23 0x7ff841e9ffa4 in g_closure_invoke (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0xffa4)
    #24 0x7ff841eb1fc0  (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x21fc0)
    #25 0x7ff841ebad5b in g_signal_emit_valist (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x2ad5b)
    #26 0x7ff841ebb08e in g_signal_emit (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x2b08e)
    #27 0x7ff84350e6ed in gtk_widget_activate (/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0+0x2496ed)
    #28 0x7ff8434098dc in gtk_menu_shell_activate_item (/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0+0x1448dc)
    #29 0x7ff843409c45  (/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0+0x144c45)
    #30 0x7ff8433f7aeb  (/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0+0x132aeb)
    #31 0x7ff841e9ffa4 in g_closure_invoke (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0xffa4)
    #32 0x7ff841eb256d  (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x2256d)
    #33 0x7ff841eba7f8 in g_signal_emit_valist (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x2a7f8)
    #34 0x7ff841ebb08e in g_signal_emit (/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x2b08e)
    #35 0x7ff84350f93b  (/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0+0x24a93b)
    #36 0x7ff8433f6283 in gtk_propagate_event (/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0+0x131283)
    #37 0x7ff8433f663a in gtk_main_do_event (/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0+0x13163a)
    #38 0x7ff84306ac8b  (/usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0+0x5ac8b)
    #39 0x7ff841bc9196 in g_main_context_dispatch (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4a196)
    #40 0x7ff841bc93ef  (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4a3ef)
    #41 0x7ff841bc9711 in g_main_loop_run (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4a711)
    #42 0x7ff8433f5696 in gtk_main (/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0+0x130696)
    #43 0x170a794 in wxGUIEventLoop::DoRun() src/gtk/evtloop.cpp:65
    #44 0x17a99b3 in wxEventLoopBase::Run() src/common/evtloopcmn.cpp:76
    #45 0x177b930 in wxAppConsoleBase::MainLoop() src/common/appbase.cpp:380
    #46 0x17f4233 in wxEntry(int&, wchar_t**) src/common/init.cpp:507
    #47 0x8547d1 in Slic3r::CLI::run(int, char**) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x8547d1)
    #48 0x8232a3 in main (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x8232a3)
    #49 0x7ff840a3882f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #50 0x84cf18 in _start (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x84cf18)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV ??:0 stl_translate_relative(stl_file*, float, float, float)
==9156==ABORTING

bubnikv, 2019-06-26:
> std::char_traits, std::allocator > const&, Slic3r::DynamicPrintConfig*,

The crash reported by @lukasmatena is unrelated, there seems to be a bug on
the current master.

lukasmatena, 2019-06-26:
I hopefully fixed the original issue with unusable extruders. The fix is in master, the issue is closed. Thanks for reporting.

The other problem with AMF loading is being solved and will be tracked elsewhere.

closed 2019-06-26 by lukasmatena