prusa3d/PrusaSlicer #2559: PRUSA Slicer 2.0.0 problem

taz43mca, 2019-06-25:
### Version
_Version of PrusaSlicer used goes here_ 2.0.0

_Use `About->About PrusaSlicer` for release versions_ 

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_ I load an STL file, it shows up on the bed, I slice it, save as Gcode; Then I click on file-new project and the bed clears and the file goes away under Name on right side of screen; then I select file- import- stl file and select a new STL file which shows up on the right under name BUT nothing shows up on the bed diagram; then I click on slice (lower right) and the file shows up on the bed; and export Gcode shows up to be selected (lower right). If I export G-Code then it works fine. 
If I close out the slicer and then open it up again, and load an  STL file, it opens with the files showing up on the bed and on the right. 
Why would I have to close out the slicer and reopen to have the file show up on the bed, or click on "slice" for the new file to have it show up on the bed?
* _Steps needed to reproduce the problem_ AFTER closing a file I have already sliced and saved G-code and selected "new project" I plan to slice a new file without closing out the slicer - I Select file-import-import stl file; the new file shows up on right panel but not on bed; I select slice now and it shows up on bed and Export G-code comes up (lower right) and I can work and save as normal.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_ Have new project open file showing it on bed.
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_no

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

taz43mca, 2019-06-25:
Additional problem with 3mf files- Open slicer, select file-import-import 3MF; file opens up and shows in right panel; BUT multiple parts sow up as/in a pile. If I select a 3MF file after opening another files and saving it and opening NEW PROJECT, I get the 3mf file showing up in right panel but the bottom right select button reads "export g-code" and not slice nor does the file show up on bed. The 3mf file is made using Windows 3D builder.

taz43mca, 2019-06-25:
Note that this piling up of the 3mf files made with MS 3d builder was also a problem with past updates of Prusa slicer.

neophyl, 2019-06-25:
Taz I think you are making a common mistake (which means that Prusa Slicer could do with being a bit clearer).

_"Then I click on file-new project and the bed clears and the file goes away under Name on right side of screen; then I select file- import- stl file and select a new STL file which shows up on the right under name BUT nothing shows up on the bed diagram; then I click on slice (lower right) and the file shows up on the bed; and export Gcode shows up to be selected (lower right). If I export G-Code then it works fine.
If I close out the slicer and then open it up again, and load an STL file, it opens with the files showing up on the bed and on the right.
Why would I have to close out the slicer and reopen to have the file show up on the bed, or click on "slice" for the new file to have it show up on the bed?"_

Have a look at the plater window, in the bottom right corner you should see 2 icons like so...
[image]

These are the 3d Editor view and the Preview.  Your models are shown in the 3d editor view and the preveiw of sliced gcode is shown in the preview.  When you load a model initially it appears in the 3d editor view.  When you click slice Plicer automatically changes the view to the preview window and you can then see your slice results.  When you export your gcode and clear it then you are clearing both views, the 3d model and preview windows.  When you then add another STL Plicer DOES NOT automatically change back to the 3dview window.  So your preview window is empty as you havent generated a preview as yet as the model has not been sliced.  

When you add the second stl file just select the 3d view icon and you will see your model.

lukasmatena, 2019-06-26:
>  Prusa Slicer could do with being a bit clearer

This has been already taken care of, see https://github.com/prusa3d/PrusaSlicer/issues/2528. Can I close the issue?

taz43mca, 2019-07-01:
You can close the first issue but still trying to figure out problem with 3mf files showing up in a pile. I have been able to seperate them one by one but unlike the stl files that appear assembled, the 3mf do not.

bubnikv, 2019-08-01:
Please attach the 3MF you are having an issue with.

taz43mca, 2019-08-01:
This is only one from many, actually I get what you will see whenever I open a 3mf file. Not with stl files.I have attached 4 files:1.. a 3mf file that was not grouped after completing2.  a 3mf file that was grouped after completing3.  an stl file that was not grouped before saving4.  an stl file that was grouped before saving
Both stl files printed OK.Hope this helps.Merrill Albury

bubnikv, 2019-08-02:
sorry no file attached

taz43mca, 2019-08-02:
Tried attaching files you asked for but this forum says it does not accept this type of files. Again. The stl files printed OK but the 3mf files open up in multiple parts. I will try and attach to a reply to your email again if the zip file does not work. Both stl and 3mf files that were either grouped or not grouped before saving. Hope this helps.

[2021 Christmas group 3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3462475/2021.Christmas.group.3mf.zip)

bubnikv, 2019-08-09:
You need to be more explicit. What is the problem? I see no difference when loading the "2021 Christmas group 3mf.3mf" or "2021 Christmas ungroup 3mf.3mf". I know the pile loaded is centered around the zero of the print bed, and I suppose PrusaSlicer just respoects the coordinates that the application that you exported this 3MF with assigned to the objects. That is, if the pile is centered around zero, then it will be centered around zero of the print bed.

taz43mca, 2019-08-09:
My 3mf file loads as multiple segments and not as one unit. The STL files load as an assembled unit. I sent two of each type because I save one as an ungrouped file and the other as a file I grouped before savings. You are correct that both open in the Prusa Slicer the same, both 3MF files are the same and both STL files open the same BUT the STL and 3MF type files do NOT open the same as described above. If you are opening these files and they are imported as one assembled unit, forgetting that they are not centered as you state in your response, then I have aproblem with my PRUSA Slicer that you do not have, and I can not filgure out why since mine was installed from the Prusa download site. I have attached a screen capture of both files
.
[image]
[image]

bubnikv, 2019-08-19:
Please try with https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.1.0-alpha1

I believe your issue has been fixed:

Fixed loading of 3MF or AMF files not generated by the PrusaSlicer. If an AMF or 3MF file does not contain PrusaSlicer's configuration, and if it looks like a multi-part object, slicer now correctly asks whether to convert it to a multi-part object #2317 #2513 #2693.

taz43mca, 2019-08-20:
You may have closed this too soon. I downloaded 2.0.0 Alfa 1 and tried the attached zipped files. The STL and 3MF files both opened larger than the saved original designs. The STL file opened all together but the 3MF file was still disjointed although it opened as one file after clicking "tes" to the pop-up box as you suggested.  The ver 2.0.0 opened the 3mf file as 5 individual files so I had to move each around to put the object back together. No pop-up box appeared. 
The last 3mf file I sent worked OK in 2.0.0 Alfa 1, it was the right size and after saying"yes" to the pop-up, it imported ok.

[IBA w I 3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3522408/IBA.w.I.3mf.zip)

bubnikv, 2019-08-21:
@enricoturri1966 would you please look into this issue? Thanks.

enricoturri1966, 2019-08-22:
@taz43mca

I took a look at your "IBA w I 3mf.zip" file.
The problem here is that this 3mf contains 4 instances of a box which all have different non-uniform scaling.
If your editor was Microsoft 3D Builder, you probably generated them by copy/pasting or duplicating the initial box you added to the model.
Unfortunately PrusaSlicer is unable to properly handle this evenience and the result is that the imported boxes are missplaced.
As a workaround for this try to add 4 distinct boxes to your model and scale/offset them as needed.
This should allow PrusaSlicer to import the generated 3mf correctly.
We will try to let PrusaSlicer smarter so that it will be able to import files as your, but this will come on one of next releases.

taz43mca, 2019-08-22:
Too bad other organizations are as responsive as you are. Thanks. I assume that the 2.0.0 Alfa 1 is still in development and not officially released yet as I could find no install files but did a work around to use it.

bubnikv, 2019-08-23:
The alpha / beta releases are avilable here:
https://github.com/prusa3d/PrusaSlicer/releases

rtyr, 2020-05-24:
It seems OK in current 2.2.0, can we close this?

taz43mca, 2020-05-24:
yes. I had not seen your last response about 3D builder, Don't know how I missed it but that did answer my question.

closed 2020-05-24 by rtyr