prusa3d/PrusaSlicer #2560: New feature request. More printer profiles for printers other than Prusa. I own a Mk.3S with MMU2S but I also own a Cr10S Pro and  Wanhao i3 plus. I would love to make Pruca Slicer my one and only slicer,  but I have great S3D profiles for them.

RogueBear, 2019-06-25:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

neophyl, 2019-06-25:
Why not create and save your own ?  I have a modified direct drive cr10s as well as my i3 and I use PrusaSlicer for both.  
Just because there aren't prebuilt ones doesn't mean you can not do it.  

Go the forum and start a thread asking if anyone is willing to share any they have already and I'll post a copy of mine and maybe others who have them will post up too.  Also I believe there are various CR10 groups that you could ask if anyone has a slicer profile.

RogueBear, 2019-06-25:
I have built my own I just think that Josef's team might do a better job than me.

bubnikv, 2019-06-26:
We simply cannot tune and maintain profiles for any 3D printer on the
planet. We will be happy to merge the profiles for other printers if they
are provided by the community. You basically need to provide a file in the
same format as
https://github.com/prusa3d/PrusaSlicer/blob/master/resources/profiles/PrusaResearch.ini

mvadu, 2019-07-02:
I agree with @bubnikv I use Prusa Slicer with a TierTime Cetus printer. But my printer has its own tweaks which are not even covered by standard gcode (see my [Prusa profile for Cetus](https://www.thingiverse.com/thing:3045287)) which I would never expect to be covered in Prusa Slicer.

DrLex0, 2019-07-17:
I have been using Slic3r, then Slic3r PE, and now PrusaSlicer, with a FlashForge Creator Pro and I've made [my own profiles and post-processing scripts](https://www.dr-lex.be/info-stuff/print3d-ffcp.html). I agree that it is not the Prusa team's responsibility to maintain profiles for non-Prusa printers, it's called ‘PrusaSlicer’ after all.
If any effort is to be spent on this, it should be to make it easier for users to get automatic updates for other profiles. There's now this vendor presets thing, but I cannot figure out how I could submit for instance my profiles into it.

bubnikv, 2019-07-19:
@vojtechkral is refactoring the support for the 3rd party profiles as of now. We are planning to support 3rd party profiles the same way Cura does, but it will take us some time to tune up the process of accepting the 3rd party profiles, testing them etc.

Deneteus, 2019-07-19:
That would seriously save people time when switching back and forth to different printers. I mean I use PrusaSlicer/Slic3r as well as as S3D and Cura and I hate having to transpose settings between programs for multiple printers out the box. You can import those settings from existing Marlin config files that have already been tested to death.

drmulligan, 2019-11-07:
I was about to make a feature request to be able to import profiles from the other major slicers but it looks like adding my opinion here would amount to the same.  Please add a feature which would allow importing my active profiles from at least Cura and Simplify3D.  If they don't quite 1-1 transpose, make it a wizard and ask the user to solve issues.

neslekkim, 2019-11-17:
I see there are commits for profile for Ender3, are we going to see support for Ender5 and Ender5 plus also?

rkarlsba, 2020-03-18:
I was wondering about CR10S as well. Ender 3 is already in the rc4 and CR10S (or similar) isn't that different.

bubnikv, 2020-03-18:
We hope for contributions of 3rd party printer profiles.

bubnikv, 2020-10-28:
@DrLex0 Are you interested in adding your FlashForge Creator Pro profiles to PrusaSlicer wizard? @rtyr may help you with that.
I understand that we would likely have to bundle the gpx utility with PrusaSlicer, which is roughly 100kB zipped. I don't think that is an issue. Cura distributes gtk utility with the X3GWriter plugin, S3D distributes it by default.

DrLex0, 2020-10-28:
@bubnikv That would certainly make things easier for the users of those profiles.
My concern is that getting these profiles to work without surprises on every platform, still requires a bit of a convoluted recipe. This can be seen from the instructions that are now in the README of [the GitHub repository](https://github.com/DrLex0/FFCP-GCodeSnippets). Especially Windows is problematic because it requires getting either a bash post-processing script to work under WSL or Cygwin, or a more limited BAT alternative of this same script (when invoking bash isn't possible).

The post-processing script is currently essential to avoid a problem when printing with only the left extruder, which is caused by inability to specify the tool number per extruder, and code in a single-extruder profile always being generated with T0 arguments (issue #2210 or Slic3r 4003). If this could be resolved, then for basic functionality it would suffice to invoke gpx with arguments `-p -r1d` as the final post-processing step.

mvadu, 2020-10-28:
@DrLex0 any reason why the scripts can't be ported to powershell? since powershell is available on Linux, Windows as well in WSL?

DrLex0, 2020-10-28:
@mvadu I have zero experience with powershell, and none of the platforms I use PrusaSlicer on, have built-in support for it. I'd rather convert the script to Perl because at least I'm familiar with that, it is available in Linux (and therefore WSL) and OS X by default, and installing a Perl runtime in Windows is pretty easy (and is by the way required for the non-WSL script anyhow).
But my point was that we could do without any script at all, if the tool number issue would be solved.

bubnikv, 2020-11-19:
@DrLex0 

> The post-processing script is currently essential to avoid a problem when printing with only the left extruder, which is caused by inability to specify the tool number per extruder, and code in a single-extruder profile always being generated with T0 arguments

How about to provide a "replace" script inside PrusaSlicer? Something like S3D
https://simplify3dwiki.com/w/index.php/Filament_change_using_REPLACE

We have the boost replace available
https://www.boost.org/doc/libs/1_53_0/libs/regex/doc/html/boost_regex/ref/regex_replace.html

How about to add a multi-line input field for multiple regexp replace commands? Would it resolve the need for an external post-processing script?

mvadu, 2020-11-20:
@bubnikv A built in replace command would be an welcome move. It may not solve @DrLex0 needs, but it will definitely help others. In my specific case, I am using a `Post Processing Script" to trigger powershell to replace `M73` with `echo` because smoothieboard I am using does not support M73, but supports echo.

```
"c:\windows\System32\WindowsPowerShell\v1.0\powershell.exe" -Command "& { (resolve-path \"$($args -join ' ')\" | get-content | ForEach-Object { if($_ -match \"^M73 P(?"<percent>"\d+) R(?"<timeleft>"\d+)$\") {Write-Output \"echo Completed:$($Matches.percent)%, $($Matches.timeleft) minutes left\"} else {write-output $_}}) | out-file -Encoding ascii $($args -join ' ')}";
```

DrLex0, 2020-11-20:
@bubnikv A replace would help if it supports backreferences, which apparently is the case.
But it would still be an ugly workaround, and the workaround also only works by the grace of GPX assuming that an M104 command without a T argument is meant for the currently active tool.
The best solution would still be the possibility to override the implicit tool number with an an explicit one in the extruder settings.

rtyr, 2021-02-02:
3rd party printer profiles are now supported, so I believe we can close this issue

https://github.com/prusa3d/PrusaSlicer/wiki/Vendor-bundles-and-updating-process

closed 2021-02-02 by rtyr