prusa3d/PrusaSlicer #2561: PrusaSlicer hangs when zooming in plater too much with perspective active.

jmd73, 2019-06-25:
### Version
version_2.0.0-233-g2f806dedc

2f806dedc762c1ee810be44820aaff8de8cb5d07

### Operating system type + version
PrusaSlicer
Version:   2.0.0
Build:     PrusaSlicer-2.0.0+UNKNOWN

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18362), 64-bit edition

OpenGL installation
GL version:   4.6.0 NVIDIA 419.35
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1070/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
[... 341 lines cut ...]
When zooming in on the backside of printbed it doesn't lock up or move toolbars/widgetbars.
The lock up since to be in the function below and specifically the while-loop, because of m_frustrum_zs getting weird values.

(Below is copy/paste from Visual Studio watch1)
▶ | m_frustrum_zs | (-2.6821498977369629e-06, 305.23953961187647) | std::pair<double,double>
-----This is the function and start of while loop (from Camera.cpp).
void Camera::apply_projection(const BoundingBoxf3& box) const
{
    m_distance = DefaultDistance;
    double w = 0.0;
    double h = 0.0;

    while (true)
    {
        m_frustrum_zs = calc_tight_frustrum_zs_around(box);
----------

* _Steps needed to reproduce the problem_
1. Start PrusaSlicer
2. Start to zoom in with mousewheel until it locks up.

* _Expected Results_
Zooming in stops before locking up.
* _Actual Results_
[image]

enricoturri1966, 2019-06-26:
Working on it -> https://dev.prusa3d.com/browse/SPE-969

jmd73, 2019-06-26:
I can confirm this is now fixed!

Thank you @enricoturri1966!

This is now ready to be closed.

closed 2019-06-26 by jmd73