prusa3d/PrusaSlicer #2562: Define/share variables in gcode sections

wavexx, 2019-06-25:
Would it be possible to define new variables in gcode sections?
Even simple expressions would really benefit from it:

```
{factor=1.5}
M221 S{(layer_z < 0.01? 100*factor: 95*factor)}
```

Shuffling this around to make it even more readable:

```
{flow=(layer_z < 0.01? 100: 95)*factor}
M221 S{flow}
```

bubnikv, 2019-06-26:
It is not a five minute job, but it is doable. What's unclear is what
should be the scope of these variables for the various custom G-code blocks
and if the scope is to be shared between blocks, then what will be the
depencencies. Not sharing the variables will be much easier for sure.

wavexx, 2019-06-27:
I would love if you could share definitions across sections (I originally intended to ask explicitly for this but later refrained). But since you asked... :)

For instance, this would allow to define some variables in the filament gcode and reuse those in the layer-change sections without having to duplicate a printer definition as I have to do now.

I think a simple global scope would be sufficient. What's more important is the order in which these variable are assigned/evaluated in order to make the sharing more useful.

I would assume these to be assigned/evaluated in the following order:

- start gcode
- filament gcode
- before/after layer change
- end

so that the filament gcode could set some vars to be used in the before/after layer change.

Sadly we don't have a "print settings" gcode section. If there would be one, I would execute this before all the others so that I could be able to set some factors which are specific to this print setting. For instance, in a fast print setting I might want to influence the filament gcode to run 1% hotter all-else being equal. Trivial to do with a multiplier, and doesn't require anything special from a UI perspective.

lukasmatena, 2024-02-18:
This is implemented since [2.6.0-alpha6]https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2024-02-18 by lukasmatena