prusa3d/PrusaSlicer #2563: Feature Req: Money multiplier

KnatteAnka, 2019-06-25:
### Version
_Version of PrusaSlicer used goes here_
2.0.0+Win64

### Operating system type + version
Win 10 Pro

### 3D printer brand / version + firmware version (if known)
MK2S FW 3.2.3

### Feature Req
all prices looks to be in EURO that is writen in but i do want to change it to show that i see as real cost in my currency.

current way is to do a copy of every filament and change the prize upp with a multiplier of 22.30

what i want is a setting to set a multiplier globaly in Configuration/preference

rapkyt, 2019-06-25:
I think the Cost input inside "Filament Settings" is in money/Kg unit, it doesn't matter the currency of it.
What PrusaSlicer does is to multiply that value with the amount of Kg needed to the sliced print.

jkavalik, 2020-04-30:
@rapkyt yep, but "money" is not configurable now and thats a problem, the official profiles can't be changed, so I cannot just modify the price to "my money", I would have to duplicate all of them just to change that one parameter and because official profiles cannot be hidden I end up with a mess.

rapkyt, 2020-04-30:
@jkavalik oh, sorry, I don't use official profiles for filaments

closed 2026-09-29 by prusaslicer-bot[bot]