prusa3d/PrusaSlicer #2564: Remove octoprint login from printer profile #featurerequest

HaasNL2, 2019-06-25:
I use the default printer profiles, but I use Octoprint.
Result is now that I have a custom printer profile just to set my Octoprint login.
Which is annoying especially when you want to update, since Octoprint login has nothing to do with the print settings I would suggest to make a separate tab for Octoprint, or save the login info separate from the printer info, and when sending the gcode ask which Octoprint server if there are more then one.

BerndJM, 2019-06-25:
Not a good idea (my opinion).
If you have more than one printer and / or different printers it is perfect as it is now. Select "Print Settings", "Filament", "Printer", "send G-code" and done. 
No risk of sending the file to the wrong printer - as with your solution would quickly be possible ...

HaasNL2, 2019-06-26:
> 
> 
> Not a good idea (my opinion).
> If you have more than one printer and / or different printers it is perfect as it is now. Select "Print Settings", "Filament", "Printer", "send G-code" and done.
> No risk of sending the file to the wrong printer - as with your solution would quickly be possible ...

But it is just as easy to select the wrong printer profile as it would be with the selecting the wrong octoprint install. I think it is even more easy to forget to set printer profile since this is done while configuration the print.
I really think it would be better to get this question when you hit send. And not when you are setting up your print. If you use the same settings for all printers you should use the same settings for all printers.
If you have multiple 2D printers you do not make your word document for the printer, but select the printer when you hit print.

BerndJM, 2019-06-26:
When I - while setting up a print - choose the wrong printer profile I will send the print job to the wrong printer, but with the correct settings for this printer.
When I set it up with the right printer and then send it to the wrong printer - that will be a mess 'cause the settings are not appropriate for the printer.

Your comparsion to 2D-printers lags a bit. You don't select the printer, you select the printer driver which takes care of the printer specific things to send with your document ...

HaasNL2, 2019-06-26:
We could still couple Octoprint login to a printer profile. Instead of a printer profile to Octoprint.
You do not put a 2d printer ip adres in your driver file, but you do hook up a driver to a ip address. If you see the printer as a driver in your analogy.

rtyr, 2021-02-02:
Shoud be solved in 2.3.0 by physical printers. Closing.

closed 2021-02-02 by rtyr