prusa3d/PrusaSlicer #2566: Preview doesn't show the parts offset by tool offset

fuho, 2019-06-25:
### Version
2.0.0+win64

### Operating system type + version
W10

### 3D printer brand / version + firmware version (if known)
Wanhao Duplicator 4S

### Behavior
[image]

Dual extruder setup; After slicing the preview shows the part shifted by the tool offset as can be seen in the screenshot. (The faded part is the shell shown in the correct position but the gcode preview is shifted by the tool offset of the 2nd extruder.
[image]

Expected result:
It would show in the center of the platform, same position as the "shell"

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[CaseV1.zip](https://github.com/prusa3d/PrusaSlicer/files/3327506/CaseV1.zip)

lukasmatena, 2019-06-26:
I can reproduce it in current master. The issue appears even in case the shifted extruder is not actually used, it is enough to just set it to default.
It appears as if https://github.com/prusa3d/PrusaSlicer/issues/1812 was resurrected. Maybe @enricoturri1966  could be interested.

enricoturri1966, 2019-06-26:
Working on it -> https://dev.prusa3d.com/browse/SPE-977

lukasmatena, 2019-07-12:
This should be fixed by https://github.com/prusa3d/PrusaSlicer/commit/2ee572bd31d38a90d87c85794c7d0a6fa71e552c, which will be part of the next release.

The problem was that Sailfish uses M135 Tn command to do a tool change. This command was not properly interpreted by the gcode analyzer so the preview showed everything as if printed with one extruder. Same for MakerWare firmware.

closed 2019-07-12 by lukasmatena