prusa3d/PrusaSlicer #2567: Easy way to comment out M203 Marlin does not use it

Wallys3d, 2019-06-26:
### Version
_Version of PrusaSlicer used goes here_ Prusa 2.0.0

_Use `About->About PrusaSlicer` for release versions_2.0.0+win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version windows 8.1 64 bit
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)Folgertech 2020 i3 
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_ Marlin

### Behavior
* _Describe the problem_ Marlin does not use M203
* _Steps needed to reproduce the problem_ slice object  safe to SD card hit print
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_ a 3d printed object
* _Actual Results_ abort print
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Wallys3d, 2019-06-26:
Thanks to bubnika i was able to find out that Marlin does not use M203, all i had to do was comment it out in the gcode all works find my problem is i forget to do this when i slice now, i really like Prusa2.0.0 does a really nice job just need a trick to get around manually having to comment out this command,I would really appreciate any help

supermerill, 2019-07-05:
Seems to be printed only with marlin flavor.
Use reprap instead?

bubnikv, 2019-08-01:
@Wallys3d did @supermerill 's advice help?

lukasmatena, 2019-10-09:
No response, but @supermerill 's advice would solve the issue. Whether this is or isn't the best way from the UI perspective can be discussed in e.g. #2755 and #2420. Closing this one.

closed 2019-10-09 by lukasmatena

ril3y, 2019-11-09:
I would still like to use marlin.  Is there a way we can make it so that we can still use marlin and not have these mcodes generated?

lukasmatena, 2020-10-16:
@ril3y Solved in 2.3.0 (alpha will be out in a few days). See #1212.