prusa3d/PrusaSlicer #2568: raft is larger than necessary

jsprenkle, 2019-06-26:
### Version
PrusaSlicer 2.0.0

### Operating system type + version
Windows 7

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk3s

### Behavior
The problem:
Model a flower pot. Print with the open side down. Add a raft to keep it on the bed.
The raft fills the entire interior of the part.
Expected Results:
Raft should only be placed near points that contact the print bed.

_Is this a new feature request?_   Yes

neophyl, 2019-06-26:
Sounds like you need to use a brim instead of a raft if you want more surface area but don't want it under the whole part.

jsprenkle, 2019-06-26:
A raft worked. A brim did not.

neophyl, 2019-06-26:
Then I don't understand the problem.  
A raft by design covers the entire area under a part, a brim goes around the edge.  Both are methods of increasing surface area and thereby bed adhesion.  If you don't want the inside of your part to included in the bed adhesion then that pretty much leaves the brim option.  Increase the brim size so that the surface area is the same as with a raft.  

btw the usual cause of poor bed adhesion at least with PLA, is surface contamination from various things, usually skin oils from touching the bed.  A thorough cleaning with basic unscented dishwashing soap (fairy liquid in the UK) and water scrubbed with paper towels until there's no streaking produces great adhesion.  I've printed quite tall (10cm high tubes that are 2 perimeters thick and about 16mm diameter) without having to use any brim or other supports since I started properly cleaning the bed.

jsprenkle, 2019-06-26:
>A raft by design covers the entire area under a part
Why waste plastic when it's not necessary? What is improved by placing a
raft long distances from any part?

> Both are methods of increasing surface area and thereby bed adhesion
Brims are not part of the object being printed. They form a ring around the
part and separate from it.
This blocks air flow to some extent, and may increase adhesion by allowing
the first layer to better 'glue' to the bed.
They do not increase the area of the *printed part* on the bed.

The issue I typically have is late in the print the head may crash into the
part or the support. The torque generated tears the print from the bed.
Brims do not help. But as I said, why waste plastic on something that does
nothing?

Perhaps I've missed some detail of the physics?

neophyl, 2019-06-26:
Whenever I use a brim it is connected to the part. It does easily removed but it is physically connected. 
It sounds from your further description of the problem that it’s isnt bed adhesion at all but curling. If it’s further up the print and the print head is hitting and tearing off then that is usually caused by the contracting plastic rising up.   Fixes for that can depend a lot of the part design itself but common advice is to increase cooling.  Not always easy but sometimes slower down the speed giving the part fan more time to cool the part sometimes works. If you are more adventurous then a different fan shroud or a better fan for cooling is a possibility. 

I’ve only had one model where this happened to me and nothing I tried  speed wise or any other software setting fixed it. I ended up modifying the design to reduce the areas that were curling up (a couple of upwards sloping overhangs). I made them straight across bridges and model printed fine. 

Sorry to not be more help but I don’t think this is a software related issue. Try the print speed reduction first though as you may have more luck than I did.

jsprenkle, 2019-06-26:
I'm not having curling issues that I can see (at the corners of the part on
the print bed). I believed it was over extrusion caused by filament
inconsistency of printer misadjustment.

Thanks for trying to help but this wasn't intended to be a request for a
solution to a problem.
It was intended to be a new feature request. Sorry if I categorized it
incorrectly.
As I said, why waste time and plastic if it's not needed.

jsprenkle, 2019-06-26:
I see the confusion. I was conflating "skirt" with "brim"

bubnikv, 2019-08-01:
A picture is worth thousand words. There is always a compromise between stability and the amount of material wasted for the support structures.

Goury, 2019-08-13:
Dear developers, can we please stop this hostile behavior and just accept the reality?

What is clear: PrusaSlicer lacks a lot of settings, including raft settings. Such as filling raft inside and distance from part edges, with separate setting for inside and outside expansion. And also raft distance setting is missing yet it was reported many times over a decade.
What is happening every time I come by: someone is looking for an excuse to not improve PrusaSlicer and to not accept a feature request.

bubnikv, 2019-08-19:
> Dear developers, can we please stop this hostile behavior and just accept
the reality?

I am very sorry if you feel offended. Please understand that if we
developers are to spend excessive time on github issue tracker, we will not
implement new features and we will not fix the old ones. It really takes a
lot of time and brain power to try to understand what you users are trying
to tell. Therefore a picture is worth thousand words.

> What is happening every time I come by: someone is looking for an excuse
to not improve PrusaSlicer and to not accept a feature request.

No. The day has 24 hours and we have our families to spend at least some
time with.

Goury, 2019-08-19:
Then how's about you ask Patric or Joseph to stop ignoring me on email/skype?

Because not only I've got a job offer from Prusa Research some (long) time ago and was ready to help you with all my skill and knowledge (and one of my skills tells me so that you have to have a separate issue tracker manager to spend time here), but I was later just ignored.

I can clearly see flaws in Prusaresearch's software development process and in project management and I know I can help, and I don't ask for too much money in return (because doing the right thing and stuff), but giving a job offer with salary number and conditions and stuff and then just ignoring every single message from me is not just insulting, it shows existence of some greater problem that I can't even imagine.

bubnikv, 2019-08-19:
@Goury Then how's about you ask Patric or Joseph to stop ignoring me on email/skype?

That is not true. You and the Prusa Research company did not meet a mutual agreement on your contract at Prusa Research. You received a response from our HR representative, to which you did not respond. That's all.

Goury, 2019-08-19:
No, @bubnikv, This is not true.
There was some questions, but I've received no response nor did I got anything but being ignored from anyone else.
Not a single message.
Nothing but complete silence.

bubnikv, 2021-02-24:
Implemented with fcb714cd241f0847522b9aac57da6b62acdd0152

New parameters raft_first_layer_density and raft_first_layer_expansion
to influence the 1st layer of raft or support.

closed 2021-02-24 by bubnikv