prusa3d/PrusaSlicer #2571: Bulges, echos, and ringing

carterm2, 2019-06-26:
### Version
2.0.0+

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version

10.14.5 Mojave

_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)

Prusa I3 Mk3 3.7.1 2266

_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
Ghosting, ringing, echos...

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/witness-lines-in-print-prusa-slicer/

and here

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/witness-lines-in-print-prusa-slicer/#post-149749

PrusaSlicer adding width to layers.  Bulges or echoes of features.  When printing parts with bosses or any type of feature, there is an echo or bulge added to the part.  See images in links provided.

Steps required... open software, add part, slice and export to printer... 

notice the image and pay attention to the shading next to the boss openings...

[image]

STL file in 2nd link above...

* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
no
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

carterm2, 2019-06-26:
3mf file attached for bulges, echos and ringing

Vue-4body.3mf.zip

tkircher, 2019-06-27:
This has been the case for years. I just sort of assumed the ghosted features were some way of adjusting for distortion.

carterm2, 2019-06-27:
What are you referring to?  What is “This”

tkircher, 2019-06-27:
http://www.learnersdictionary.com/definition/this

carterm2, 2019-06-27:
Issue resolved...

This is a settings solution.  The out of box settings for printing are causing defects and likely many users will be put off the software until it is fixed.   

Please see this link for solution... 

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/witness-lines-in-print-prusa-slicer/paged/2/#post-149888

Close this issue... 

Thx

Carter

closed 2019-06-27 by carterm2