prusa3d/PrusaSlicer #2574: Sliced objects moved off build plate

richardsm32, 2019-06-27:
### Version
2.0.0

### Operating system type + version
Win 10 Pro
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3

### Behavior
*_Sliced objects occupy large base area. One of them is partially OFF the build plate. There is no layer shifting invovled_
* _Sliced objects are close the edges in Prusaslicer but with about .5 cm tolerance on each side_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_*INCLUDED*
[image]
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
[Player_Tray_v2.zip](https://github.com/prusa3d/PrusaSlicer/files/3333723/Player_Tray_v2.zip)

lukasmatena, 2019-07-15:
I opened your 3MF, sliced it as it was and tried to print it. The result seems to almost exactly match the 3D Preview. This would indicate that PrusaSlicer does the job correctly and there is something wrong on your end. Can you reproduce the issue? Does/did it happen with other gcodes, or only this one?

richardsm32, 2019-07-15:
So far this has been the only gcode where I've experienced this issue. All
my other prints have turned out great, including all those sliced by
Prusaslicer 2.0. I can try the same gcode again when I am home to see if it
happens again. Perhaps my build plate wasn't squared properly.

lukasmatena, 2019-07-15:
> Perhaps my build plate wasn't squared properly.

I would guess it was an incorrect homing. If, for example, something prevented the x-carriage to travel to the very left during homing, the result would have been exactly as you show (the printer would think the origin is a bit to the right than the expected position). Anyway, close the issue unless you're able to reproduce it.

closed 2019-07-23 by richardsm32