prusa3d/PrusaSlicer #2575: Droop compensation for horizontal holes

haarp, 2019-06-27:
PrusaSlicer 2.0.0, this is a feature request.

Overhangs are a challenge to print, but that doesn't mean that they can't be improved. When printing curved overhangs, like holes oriented horizontally, the top of the hole tends to droop down somewhat, distorting the shape. This looks bad and can be problematic for example if one intends to feed a bolt through with tight tolerances.

The well-known Benchy has a few of these kinds of overhangs. Here's an example:

[image]

To combat this, some designers modify their holes from perfectly round to slightly teardrop-shaped. Let's call this droop compensation. For example, the Bear X-end:

[image]

This greatly improves the shape of the hole at the top. It won't be perfect of course, but perfect isn't possible with a hole in this orientation anyway. At the very least, it makes it easier to feed a round bolt through.

I'd like to suggest that PrusaSlicer gain a droop compensation as a print setting. It should be the slicer's job to account for the minutiae of the technology, and such is much better suited to deal with this than the part designer. Plus it would allow the compensation to be applied to existing parts that do not use teardrop compensation, or disabled for parts that do.

Thanks!

supermerill, 2019-07-04:
"It should be the slicer's job to account for the minutiae of the print technology"
meh
"is much better suited to deal with this than the part designer"
clear no. The slicer don't know anything about the part designer intent & thinking.

Also this feature is extremely difficult to code and/or may have many many side-effects.

rennne, 2019-07-12:
The part designer has sometimes no influence, on the orientation in which the print will be done later on.

Also when downloading an stl file, it is quite difficult for most people to get this feature into an existing hole.

supermerill, 2019-07-12:
problem is, how to distinguish overhangs/bridges that should be cut to create a "teardrop-shaped" and the ones that should stay intact?

haarp, 2019-07-12:
> problem is, how to distinguish overhangs/bridges that should be cut to create a "teardrop-shaped" and the ones that should stay intact?

You don't need to. You make it a setting in the slicer, then people can either enable it globally, or add modifiers to the part in Prusaslicer and enable the setting for where it's needed.

rvt, 2019-07-13:
> (i'm just a random guy on internet that make code for some slicers)

That does´t mean you are right...

I do believe it´s the slicer´s job to understand in what material and how a object is printed. There are many tool path generators far outside the realm of amateurs and even professionals that can calculate and advice on some designs and do understand the device and materials used in the process. For example injection mold design applications or surface modeling.

I am not saying it´s easy to solve the reporters issue, I am just saying it´s not a clear ´no´ and there is room for slicers to improve on this area.

supermerill, 2019-07-15:
I  just put that to be sure to not be confused with the prusa team.

fthouret, 2020-05-05:
Hi !

Just to come back on this. Why not a specific modifier to ask PrusaSlicer to add this at the top of a inner hole. Even if we have to do this to every hole, it should be usefull.

adamdport, 2021-02-20:
@fthouret I agree, if we can now paint on seam and support blockers/enforcers, it seems like painting on droop compensation is totally viable. Would love to see this!

> > problem is, how to distinguish overhangs/bridges that should be cut to create a "teardrop-shaped" and the ones that should stay intact?

@supermerill they're not really "intact" as is (eg. bolts don't fit), but I do see your point. Maybe a simple tradeoff could be either a "maximum overhang distance" or "maximum vertical angle" that we can set? This could get really wonky if you have a model with major features beyond those angles, but if you stuff them away in the advanced menu then I think it could be low hanging fruit for now?[

haleypearse, 2022-09-28:
Howdy, I'm the author of the above feature request closed as duplicate. I'm curious whether there's a good reason my solution isn't good. (the request is a bit different in being for all unrealistic slopes, not just holes)

> I'm thinking there must be a point at which perimeters are generated by finding the model's intersection with a horizontal plane at a given z. I'm suggesting at this point also generating a bounding perimeter that is an outer offset from the previous layer's perimeter, according to an angle parameter and the layer height. This bounding perimeter being intersected with the nascent perimeter would accomplish the task, before other print parameters are accounted for.

DragonFinlay, 2022-10-04:
I've been reading through this and #3817 and I find your ideas really interesting. I'll remind the rest of the team about this and see what they say. I think especially the "paint-on overhang compensation" could be something that's possible to implement.

Although I'm not a programmer/coder and I have no idea about the actual scope of implementing something like this. If it's as bad as @supermerill suggests, it might take a "little" while.

closed 2026-09-29 by prusaslicer-bot[bot]