prusa3d/PrusaSlicer #2577: Copying Temporary G-code Error

leroythedog, 2019-06-28:
### Version
_Version 2.0.0

### Operating system type + version
IOS 10.14.5

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version - Prusa 2.5S / MMU2S

### Behavior
Has been working fine, but last 2 days.  Crashes when trying to save gcode.  
See Pic.
[image]

_Is this a new feature request?_
Yes.
I have tried deleting PrusaSlicer and reinstalling, but problem persists.

bubnikv, 2019-07-01:
Are you exporting the G-code to a SD card? Or sending to OctoPrint? Is your
SD card read only? Is your SD card full?

leroythedog, 2019-07-01:
Thanks for your response.  I found assistance from the facebook forum.  It was my SD card.  Even though it wasn’t full, it decided to tell me it was full and didn’t accept any more files.   I have switched to a new SD card and things are now better.  

Joe O.

closed 2019-07-01 by vojtechkral