prusa3d/PrusaSlicer #2578: Feature Request: Save Object Layout/Modifiers to G-Code

chezy525, 2019-06-28:
### Version
2.0.0 + win64

### Feature Request:
Include a listing of object/instances to exported G-Code, including specific layout location/rotation/scaling and any modifiers to the base settings.  I think this would fit before or after the existing settings that are saved at the end of the G-Code file.

### Reasoning:
I really like the concept of being able to save a project, but I find the current implementation to provide me no benefit.  In short, I frequently find myself making adjustments to the actual design files of the things I've printed to make adjustments in future prints.  As such, with the current setup, after making any changes I then have to re-add the (new) objects to the plate, then re-add any of the modifiers.  Getting everything to the same location, and making all of the same modifications is often nearly impossible (especially with Variable Layer Heights!).  Saving all of these details and allowing me to change/update the underlying STL files, is really what I want from a "saved project", however I completely understand that there is a lot here and it's not a simple change.  So, I think that adding most (if not all) of this information as comments at the end of a G-Code file is a good mid-term compromise that doesn't require everything to be completed before some of it is available and useful.  I'd still need to manually enter the values, but they'd at least be saved in a way that I could read them.

bubnikv, 2019-07-01:
We are not going to add more Slicer data to the G-code. You should save
your data to a project (File -> Save Project), the project file will
contain all the modifiers with their properties.

bubnikv, 2019-08-01:
Modifier data will not be stored into the G-code. Closing.

closed 2019-08-01 by bubnikv