prusa3d/PrusaSlicer #2579: Slicing helical screw

standarch, 2019-06-28:
### Version
2.0.0 + Linux 64

### Operating system type + version

Ubuntu 18.04.2 LTS

### 3D printer brand / version + firmware version (if known)

Prusa Mk3s

### Behavior
* _Describe the problem_

Not drawing a path for part of the object

* _Steps needed to reproduce the problem_

Try slicing the attached file

* _Expected Results_

A helical screw
* _Actual Results_

most of a helical screw, but part missing, and some extruded material misplaced within prusa slicer.
This sliced object has been cut away so that it easier to see how it was sliced. That is the top part of the object was sliced as well, but is not shown, because the slider was adjusted.  [See second picture of sliced object, slider control on rhs of image]

[image]

_Is this a new feature request?_
probably not

[image]
[image]

[helixTest1c.zip](https://github.com/prusa3d/PrusaSlicer/files/3341155/helixTest1c.zip)
[helixTest1c_3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3341157/helixTest1c_3mf.zip)

haarp, 2019-07-04:
You can try enabling "Detect thin walls". But yes, PrusaSlicer struggles a lot with features that are thinner than one extrusion width.

supermerill, 2019-07-05:
It's not even that, the model isn't slice correctly.

Jebtrix, 2019-07-05:
That model mesh is non-manifold. Slicers attempt to automatically fix this but its a best effort solution, not a perfect science. If you have ever tried to manually fix this mesh problem (ie in Blender) you would understand how problematic it is. To visualize this here is pic of all the non-manifold edges of that model highlighted.
[image]

Cura's attempt at repair does yield better results but if you examine it more closely it is still not correct. The center should be a hole buts its being filled in depending on infill percentage (picture is 15%). 
[image]

If you rotate the part vertically and inspect the slice you can see its auto-repair is causing the center hole to be half filled in a spiral pattern. 
[image]

Auto-repair is never perfect and I'm actually amazed that Slicers manage to print some truly garbage meshes.

Jebtrix, 2019-07-05:
Just want to add this part still could still be a useful test object to improve PrusaSlicer's auto-repair. I just find it hard to fault Slicers with bad meshes.

standarch, 2019-07-08:
Thanks guys appreciate your help. By the way Jebtrix, did you use blender to show the manifold errors?  Are the normals to the constitutent triangles facing in the wrong direction?  I had a look in blender a while ago.  Blender might be able to correct it automatically.  I think I tried doing that, but its been a while now, and I forget.  I think its auto correction didn't work or maybe I'm not familiar enough with Blender.

 Eventually I switched to Design Spark Mechanical from FreeCad which generated the correct mesh. DSM seems a bit more reliable than Freecad, but who knows what will happen to it. Its not open source, but I think RS will keep it free to use because of their business model. Pity it needs to have a windows system to run it. 

[Helix working amf gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/3370543/Helix.working.amf.gcode.zip)

Jebtrix, 2019-07-09:
I used opensource [MeshLab ](https://github.com/cnr-isti-vclab/meshlab) for that picture. That mesh has Non Contiguous edges and Multiple Faces. In Blender if you use Select Non Manifiold in Edit Mode, it will highlight all non-manifold "everything". There is a tool panel that will appear that lets you check/uncheck Multiple Faces, Non Contiguous, or Vertices problems to isolate specific problems. There really isn't an auto-fix in blender. You can sometimes get away with Remove Doubles and Recalculate Normals to get rid of some problem areas. Details of fixing meshes is out of the the scope of a Github issue (Learn some blender, steep learning curve but a very useful tool to have in your arsenal).

Here I bisected it so you can see the mess of Multiple Faces (edges shared by 3+ faces) and extraneous geometry. Visually the solution here looks "easy" to the human eye only because we can infer intent but that has to be algorithmically challenging.
[image]

I removed the center hole from the mesh but fixed nothing else. I then sliced with PrusaSlicer (Cura gave similar results).  This simplifies the auto-fix results so you may understand a little better what is actually happening. 
[image]

You see how the the spiral is essentially detached? The auto-repair detected a Multiple Face edge. In this case it has 4 faces to one edge. To make it manifold it has to choose only 2 faces for the edge. For the other faces it must do an edge split (duplicate) that only connects to 2 other faces. I tried to illustrate that here:
[image]

Some people assume non-manifold auto-repair would automatically fill this resulting cavity in. This is where some have strange missing infill and other slice oddities when the model visually looks fine (from outside).

standarch, 2019-07-16:
Thanks Jebtrix.  It was created merging two objects.  The outer helix, and a solid cylinder.  The intersection of the two objects is the space labeled "EMPTY" in your diagram.  Its a pity freecad couldn't identify the inside of the cylinder, and not add any surfaces within it, when it did the merge.
Yes, I thought this 3d printing was going to be easy.  The printing is fairly simple, the cad ... not so much.  The slicer is not particularly at fault.  The cad drafting tools I am coming to the conclusion need more work to eliminate bugs.  Freecad is a little too buggy, DesignSpark is good, but still has some bugs, especially related to doing this merge of two objects and a helix.  Actually pruser slicer did the merge of two separate objects successfully where DesignSpark did not. But this is a gross generalisation, I need to do more work to become familiar with the cad tools, and this unfortunately takes more than a couple of weeks. I would still recommend using DesignSpark it may just be unfamiliarity with the software.  It is difficult to tell sometimes whether it is a bug or a user error, when one isn't familiar with the software.
Anyway, thanks a lot for your insight in the matter, and I hope I will have some time to use blender, it looks like a relatively bug free program.

closed 2019-07-16 by standarch