prusa3d/PrusaSlicer #2580: locale 'en_US' cannot be set

IronicBadger, 2019-06-29:
[image]

Running on Arch Linux. v2.0.0.

BerndJM, 2019-06-29:
I would say the locale en_US is missing on your system.
Have a look here:
https://wiki.archlinux.org/index.php/Locale
how to check and (igf necessary) generate a missing locale.

IronicBadger, 2019-06-30:
My locale is set to en_GB thus

```
locale -a
C
POSIX
en_GB.utf8
```

If I modify it to `en_US` the error goes away. Bug?

mavelli, 2019-07-14:
i came across the same issue (archlinux, prusa-slicer 2.0.0). i would say, the prusa-slicer should offer all locales it has translations for but warn (second column in the menu) for missing locales in the system. makes sense?

mavelli, 2019-07-14:
for me, it was even more bizarre: 
i have de_CH.utf8 as default and when i installed the package prusa-slicer (2.0.0) it worked fine in an German-English mixed GUI. 
since i am new to 3d printing (my i3mk3s should be delivered next week), i decided to switch the prusa-slicer to english to follow easier the awesome videos by Josef Průša and others on the internet without the need to search for which menu means what. when tried to switch to English, the restart showed the "cannot set locale "en_US". so i tried to switch back to German - which results now the program showing "cannot set locale "de_DE" at each start (but the GUI switched to the German-English mix).

bubnikv, 2019-08-23:
There are two issues:

First, you really need to have the locales data installed and generated correctly on your system. The only thing PrusaSlicer can do is to give you an advice what to do and not crash.

Second, we may implement alternate selection of language dictionaries the same way we now redirect de_CH to German Prusa3D web pages. We will likely do that for the final 2.1.0 release.

mavelli, 2019-08-24:
i think the main issue is prusa slicer 2.0.0 GUI offering a Language selection while under the hood it expects a specific locale to be present for a specific language setting. 

since languages are used in several geographic regions with different locales, this way of switching language of the GUI will fail unless all locales are mapped to a language and the software tries to find out which locales are present. not sure if this is the best way.

in a correctly set up system, you can run "locale -a" and get all the installed locales on that system. prusa-slicer can get this information from the system at runtime and accordingly offer possible languages for the GUI and other languages with missing locales of the system can be greyed out or not shown at all. 

if you _require_ explicitely en_US and de_DE to be present on the system just to be able to select English or German in the GUI, you discriminate against all other cultures and regions also speaking German and English!
i personally do not have en_US but en_GB and en_IE on my system for english, as i am closer to these cultures language-wise - and all software i run on my machines is fine with that - fallback for "older" software is locale "C". 
as i am swiss, i have de_CH and fr_CH installed for German and French and i do not feel the need nor spending the resources (megabytes and a few seconds rebuilding time each recompile/update today are not much, but having something you do not need is against the KISS principle) to keep locales i do not use otherwise solely to have prusa-slicer be happy with its need for specific locales is a bit ... strange.

bubnikv, 2019-08-31:
@mavelli It took me around 2 and half working days and some iterations to implement what I consider sane handling of languages, and there still may be some dragons hidden in there. The changes will be part of the PrusaSlicer 2.1.0-beta3.

We have roughly 80% installation base on Windows, 10% on OSX and 10% on Linux. The most problems are reported for Linux, then Windows. This shows how Linux is fragmented and how the configuration is difficult and / or that the users often do not understand configuration of their system. While I am am a fan of the open source movement, from the product perspective supporting Linux is a PITA and some 3D printer manufacturers (for example Formlabs) do not support Linux at all.

We have changed PrusaSlicer, so that it checks on start up, whether the current locale (from the environment) is correctly installed. If not, then an error message 
"You may need to reconfigure the missing locales, likely by running the locale-gen and dpkg-reconfigure locales commands."
is produced and the application is closed.

Once the application starts, it select the dictionary matching the locale language if possible. If the language is right-to-left, the locale is switched to English. If your language is German / Swiss, the German dictionary is used while the German / Swiss locale is maintained. I believe this fits your request.

When switching the language using the language selection dialog, and the locale required by the dictionary is not available on your system (this is most likely just a Linux issue), the switching is aborted with
"You may need to reconfigure the missing locales, likely by running the locale-gen and dpkg-reconfigure locales commands."
error message.

When switching the dictionary back to a dictionary which matches your system locales, the system locales are used. Otherwise the locale paired to the dictionary language is used (and therefore mandated on Linux to be installed). The pairing of the dictionary with the language & country code is hard coded inside wxWidgets UI library. I suppose we may be a little bit smarter and we may try to find a pair of language & country that is installed on the system. That would make the code even more complex and hairy.

On Windows and OSX, the logic is even a bit more complex, as those two systems support independent settings of the language used for presentation, and the locales. Also Windows and OSX support a concept of "preferred languages", and wxWidgets (the UI toolkit that PrusaSlicer uses) provide interface to this "preferred languages" API. For example, I have my Windows set to English for presentation, while I have my locales on the same machine set to Czech. PrusaSlicer correctly starts with an English dictionary now.

The logic is quite complex and hairy and I am pretty sure there are some corner cases that are not covered. I am quite sure it will take yet quite some iterations before everyone gets satisfied.

Closing for now, the changes will be in the PrusaSlicer 2.1.0-beta3.

closed 2019-08-31 by bubnikv

mavelli, 2019-09-02:
thank you for fixing this annoying bug in PS! my condolences fighting with wxWidgets UI lib for this - not an easy task. 
regarding the only 10 % linux userbase (where do you have this metrics from?): if the prusa printers would use proprietary closed source software and not run on FOSS software, i would refuse to use it by principle. it has something to do about freedom and transparency of code - you know... thinkiering... making.. creating the future :)

bubnikv, 2020-12-05:
> regarding the only 10 % linux userbase

It is getting lower every day, Macs are killing Linux desktop, and rightfully so given the fragmentation of Linux distros and Linux desktop. You guys are doing it to yourself. I used to be a Linux militant evangelist in 90's...

We have also added separate US English and GB English choice in the language switching dialog in PrusaSlicer 2.3.0-beta2, see #3368

> in a correctly set up system, you can run "locale -a" and get all the installed locales on that system. prusa-slicer can get this information from the system at runtime and accordingly offer [...]

Thanks for the hint. Yes, we can do that in PrusaSlicer 2.4 if you guys find the current behavior insufficient, however I would rather invest our time elsewhere. For now the user has the choice of us_EN, us_GB, and his active locale is respected as well if it matches the dictionary.