prusa3d/PrusaSlicer #2581: Repeatable coredump when changing extruder of subpart

jasontibbitts, 2019-06-29:
### Version
2.0.0 from Fedora packages

### Operating system type + version
Linux, Fedora 30, x86_64.

### 3D printer brand / version + firmware version (if known)
Prusa Mk3s MMU2s

### Behavior
If I load the attached 3mf file and try to change the extruder one of the subparts, I get a segfault.  (If you don't have background processing on, the crash happens when you slice.)  To create that file I loaded an amf file, then loaded a second as a subpart of the first.  I don't know how the first subpart ended up without a name.  I can supply the two amf files that you can load in manually if that would be useful.

When it crashes, this is dumped to the console:
```
/usr/include/c++/9/bits/stl_vector.h:1060: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) const [with _Tp = Slic3r::WipeTower::ToolChangeResult; _Alloc = std::allocator<Slic3r::WipeTower::ToolChangeResult>; std::vector<_Tp, _Alloc>::const_reference = const Slic3r::WipeTower::ToolChangeResult&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
```
I suspect this issue might be unique to Fedora because by default we compile all of those assertions in, but this still looks like an OOB array reference.

I can get a backtrace with coredumpctl, but because of the symbol mangling I'm not sure how useful it is. Here is the trace for the thread that aborts; if there's something I can pass to gdb to get more useful information, please let me know.
```
Stack trace of thread 21379:
#0  0x00007f71f25cee75 raise (libc.so.6)
#1  0x00007f71f25b9895 abort (libc.so.6)
#2  0x000055a77b9db3a8 _ZSt20__replacement_assertPKciS0_S0_ (prusa-slicer.wrapped)
#3  0x000055a77bc4df8f _ZN6Slic3r20WipeTowerIntegration11tool_changeB5cxx11ERNS_5GCodeEib (prusa-slicer.wrapped)
#4  0x000055a77bc52496 _ZN6Slic3r5GCode13process_layerEP8_IO_FILERKNS_5PrintERKSt6vectorINS0_12LayerToPrintESaIS7>
#5  0x000055a77bc5a4d2 _ZN6Slic3r5GCode10_do_exportERNS_5PrintEP8_IO_FILE (prusa-slicer.wrapped)
#6  0x000055a77bc5ca5e _ZN6Slic3r5GCode9do_exportEPNS_5PrintEPKcPNS_16GCodePreviewDataE (prusa-slicer.wrapped)
#7  0x000055a77bab58de _ZN6Slic3r5Print12export_gcodeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEPNS_1>
#8  0x000055a77be4ee1d _ZN6Slic3r24BackgroundSlicingProcess11process_fffEv (prusa-slicer.wrapped)
#9  0x000055a77be4f808 _ZN6Slic3r24BackgroundSlicingProcess11thread_procEv (prusa-slicer.wrapped)
#10 0x000055a77be4faca _ZN6Slic3r24BackgroundSlicingProcess16thread_proc_safeEv (prusa-slicer.wrapped)
#11 0x00007f71f3c6c474 n/a (libstdc++.so.6)
#12 0x00007f71f27655a2 start_thread (libpthread.so.0)
#13 0x00007f71f2692303 __clone (libc.so.6)
```

#### Project File (.3MF) where problem occurs
[bug.zip](https://github.com/prusa3d/PrusaSlicer/files/3342094/bug.zip)

bubnikv, 2019-08-01:
Thanks for heads up. I will let @lukasmatena look into it, he runs his builds with ASAN enabled. He already fixed some wipe tower issues, maybe he already fixed this issue on master.

lukasmatena, 2019-08-01:
Sadly, I didn't. I added an exception throw with https://github.com/prusa3d/PrusaSlicer/commit/eb29c3e01d3ff188db74faa567ece26cc3425fae to prevent hard crashes in similar situations, but did not fix the cause. It is on my list.

lukasmatena, 2019-08-01:
Thanks for reporting, I can reproduce the problem. Hopefully we'll be able to fix it.

lukasmatena, 2019-08-05:
I hope it is fixed by https://github.com/prusa3d/PrusaSlicer/commit/6ab1cec48c037a4f4a132a06e83f2a12fcacd9ff. The reason it crashed was that your object is standing on an edge and the first layer of the print was empty (so it was not printable). An error message is now emitted before this happens. Thanks for the report.

bubnikv, 2019-08-05:
An error will be reported for such an object by the upcoming PrusaSlicer 2.1.0-alpha1. Ideally slicer would lower the object down to touch the build plate, but that change is much more involved and risky to do for the 2.1.0 release.

lukasmatena, 2019-10-01:
The issue is fixed - closing.

closed 2019-10-01 by lukasmatena