prusa3d/PrusaSlicer #2582: [Feature Requesest] Adding setting to several objects in one go

YD95, 2019-06-30:
### Version
2.0.0+win64
**Build**
PrusaSlicer-2.0.0+win64-201905201652

### Operating system type + version
Windows 10 64-bit edition

### 3D printer brand / version + firmware version (if known)
 Original Prusa i3 MK3S MMU2S / FW-i3: 3.7.1, FW-MMU: 1.0.5

### Feature Request
As mentioned in the title, it would be nice to have a feature where you could select several object and add/modify their setting like extrusion width or in my case Extruder number for Parameters, infill, ...

The motivation behind it is when you have a few objects, it's fine to manually adjust the settings per object, but when you have +10 objects then it gets tedious.

#### Project File (.3MF) as en example
Here is an example what I mean. Each one of the objects were set by hand.
The end goal of this is that I can choose some of the plates and give it different settings, instead selecting one at a time to change/add settings.
[MakerClock_Clock.zip](https://github.com/prusa3d/PrusaSlicer/files/3342833/MakerClock_Clock.zip)

bubnikv, 2019-07-01:
Thanks for your suggestion. As of now you can copy / paste objects and
modifiers including their settings. We consider adding copy / paste for
parameter groups. Your request certainly makes sense, but it is more
involved.

M0o0P, 2019-12-24:
I concure with the OP. When i import parts for MMU printing, I have to assign the extruders individually for each part. It would be helpfull to be able to select all parts that get extruder 5 and then click on extruder 5 one time rather than selecting say 10 of 20 parts individually setting the extruder each time (more prone to errors this way).

closed 2026-09-29 by prusaslicer-bot[bot]