prusa3d/PrusaSlicer #2583: Possible slicing errors in PS when in "detect thin walls" mode

mkoistinen, 2019-06-30:
(Possibly related to #227 and #149, but perhaps this will provide additional clues to the core of the issue?)

### Version
PrusaSlicer 2.0.0 - 201905201652

### Operating system type + version
MacOS 10.12.6

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S, latest firmware

### Behavior
I designed a part in Fusion 360. It consists of the primary body and additional bodies which are custom supports. These are pretty thin (0.405 mm) so I have to enable "Detect thin walls" in the Print settings. When I merge the bodies together into a single part before exporting to STL, it slices fine in PrusaSlicer and prints beautifully.

However, if I export the separate bodies as a single STL from F360, it loads perfectly fine into PrusaSlicer and displays perfectly in the Platter view (see 1st screengrab). However, when I slice it, some tiny fragments of just 3 of 20 of the custom support bodies is just missing. (see 2nd and 3rd screengrabs).

I have also tried the "Repair STL file..." in PrusaSlicer. It claims to have fixed the STL (saved as an Obj), but it makes zero difference. Obviously, the workaround is to export everything as a single body from Fusion 360, but it is not ideal.

* _Expected Results_

I was expecting no difference between exporting as a single body, vs. as separate bodies before slicing.

* _Actual Results_

Inexplicable bits are missing from the sliced part if the STL was exported from multiple bodies from F360.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

[image]
[image]
[image]

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
The 3MF file and the single body STL file can be found here: https://www.prusaprinters.org/prints/3873-gopro-hero-7-cozy.

The multi-body STL file is here for your inspection: 
[Mount w:Separate Supports.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/3343053/Mount.w.Separate.Supports.stl.zip)

Jebtrix, 2019-07-03:
PrusaSlicer 2.0.0 - 201905201652
Win7 Pro 64bit

The .3MF file slices just fine for me. MacOS related in some way maybe?

mkoistinen, 2019-07-09:
The plot thickens...

closed 2026-09-29 by prusaslicer-bot[bot]