prusa3d/PrusaSlicer #2584: Crash on startup after building from source - Win, x64, Release

BenjaminVerdier, 2019-07-01:
### Version

tag: version_2.0.0

### Operating system type + version

Windows 10 Pro v1803

### Behavior

PrusaSlic3r crashes on startup in Release mode after building from source. Built using cmake, Visual Studio 12 Win64, prebuilt dependencies as described in the [documentation](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md).

It runs perfectly fine in Debug mode.

I found that it fails at:
``` c++
HINSTANCE hInstance_Slic3r = LoadLibraryExW(path_to_slic3r, nullptr, 0);
	if (hInstance_Slic3r == nullptr) {
		printf("PrusaSlicer.dll was not loaded\n");
		return -1;
	}
```
in `src/PrusaSlicer_app_msvc.cpp`, lines 276-280

I checked, the referenced .dll file exists but hInstance_Slic3r is null. In Debug mode it finds and loads the .dll with no issues. In Release mode, it finds and loads the .dll and immediately unloads it. Output while stepping `HINSTANCE hInstance_Slic3r = LoadLibraryExW(path_to_slic3r, nullptr, 0);` :
```
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\nvspcap64.dll'. Cannot find or open the PDB file.
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\winsta.dll'. Cannot find or open the PDB file.
'prusa-slicer.exe' (Win32): Loaded 'C:\xxx\PrusaSlic3rSources\build\src\Release\PrusaSlicer.dll'. Symbols loaded.
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\comdlg32.dll'. Cannot find or open the PDB file.
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\WinSxS\amd64_microsoft.windows.common-controls_6595b64144ccf1df_6.0.17134.829_none_fb46a5473061b9d5\comctl32.dll'. Cannot find or open the PDB file.
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\wsock32.dll'. Cannot find or open the PDB file.
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\oleacc.dll'. Cannot find or open the PDB file.
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\ws2_32.dll'. Cannot find or open the PDB file.
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\ws2_32.dll'. Cannot find or open the PDB file.
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\ws2_32.dll'
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\winspool.drv'. Cannot find or open the PDB file.
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\msimg32.dll'. Cannot find or open the PDB file.
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\propsys.dll'. Cannot find or open the PDB file.
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\IPHLPAPI.DLL'. Cannot find or open the PDB file.
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\bcrypt.dll'. Cannot find or open the PDB file.
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\oleacc.dll'
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\wsock32.dll'
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\propsys.dll'
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\IPHLPAPI.DLL'
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\bcrypt.dll'
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\winspool.drv'
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\WinSxS\amd64_microsoft.windows.common-controls_6595b64144ccf1df_6.0.17134.829_none_fb46a5473061b9d5\comctl32.dll'
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\comdlg32.dll'
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\ws2_32.dll'
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\msimg32.dll'
'prusa-slicer.exe' (Win32): Unloaded 'C:\xxx\PrusaSlic3rSources\build\src\Release\PrusaSlicer.dll'
```

Any ideas?

bubnikv, 2019-07-02:
I would recommend to check dependencies of your compiled PrusaSlicer.dll
with  http://www.dependencywalker.com/
I suppose it cannot load some of the depending DLLs. Unfortunately these
missing DLLs are not reported. You may also try to track the failing system
calls using sysinternals filemon.

BenjaminVerdier, 2019-07-03:
Thanks for the suggestion. I opened PrusaSlicer.dll with Dependency Walker, and there's indeed a bunch of missing modules, a whole lot of API-MS-WIN-*.DLL and EXT-MS-WIN-*.DLL, as well as EMCLIENT.DLL, IESHIMS.DLL and LIBCURL.DLL.
I opened the Debug one as well, for comparison, and the only difference is the LIBCURL.DLL that just isn't present.

Is there a way to fix it? Is it possible to drop CURL entirely in Release?

supermerill, 2019-07-05:
I advise to build the deps, curl is inside.

BenjaminVerdier, 2019-07-09:
I built the dependencies and the exact same thing happens.
I also checked the x64 dependencies downloaded from the documentation page, and curl is in there too.

bubnikv, 2019-08-01:
There shouldn't be a dependency to LIBCURL.DLL as we link against libcurl.dll statically on Windows. It may be, that you have a dynamic version of libcurl on your system somewhere and it is picked instead of the static version compiled in slicer's dependencies? Anyway, if you find the libcurl.dll on your system and copy it next to prusa-slicer.exe, it should load.