prusa3d/PrusaSlicer #2585: Unexplained high speeds next to solid infill on Ripple Test cube

haarp, 2019-07-01:
PrusaSlicer 2.0.0 on Gentoo Linux

Hello,

for some reason, PrusaSlicer increases print speed/volumetric speed for layers which also happen to contain solid infill. The perimeters are being printed noticeably faster. An example of this are the tops and bottoms of a Ripple test cube. I'm seeing this across all print profiles.

Screenshot: [image]

Project: [Ripple_test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3347406/Ripple_test.3mf.zip)

Is this weird or am I missing something?

Cheers!

bubnikv, 2019-07-02:
This is an effect of the slow down logic defined at the filament profile.
If you switch it off, it should be all the same color. Where there is more
solid infill, the print takes longer, therefore less slow down is required.

haarp, 2019-07-02:
> This is an effect of the slow down logic defined at the filament profile.

Ah, that's what's reponsible! Thanks. Ok, then this is expected behavior. Closing.

closed 2019-07-02 by haarp