prusa3d/PrusaSlicer #2586: Trouble with PrusaSlic3r 2.0.0

TheWebExpert1, 2019-07-02:
### Version
2.0.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
i3mk3s v. 3.7.1-2266

### Behavior
If PrusaSlicer (PS) isn't the first program run, it will crash windows.
Run ANY other software.  Run PS.  It may crash immediately.  Or, it will crash when you try to slice an object.

[https://ibb.co/0FvKm49](url)

BirkBinnard, 2019-07-02:
Your experience is unique. Many of us have been running 2.0.0 on Win10 for weeks with no problems. I'd suggest checking the Windows/System Event Log for any entries that are flagged as critical. Also, try removing and reinstalling 2.0.0. There is something unique/unusual about your system that is causing your problem and it may be tricky identifying what it is.

bubnikv, 2019-07-02:
You may try to delete the configuration directory. The location of the configuration directory is described here:
https://github.com/prusa3d/PrusaSlicer-settings

Also you may want to re-install / update your graphics driver.

TheWebExpert1, 2019-07-02:
I updated the video driver and it seems to be working better. It did a
weird video glitch during slicing, but didn't crash.

Perhaps there might be some kind of issue between PrusaSlicer and the
Nvidia card

bubnikv, 2019-08-01:
Solved by reinstalling the GPU driver. Closing./

closed 2019-08-01 by bubnikv