prusa3d/PrusaSlicer #2587: What happened to Layer Editing?

xoxota99, 2019-07-02:
Using PRuseSlicer on Mac. In Slic3r PE there used to be an interactive layer editing UI, where I could set custom layer heights. This seems to have disappeared (or I can't find it any more). Where did it go?

jotomo, 2019-07-02:
I think it was moved around in the UI. Select a part to make the right-most action in the toolbar at the top middle active and the layer editing control appears on the right side (assuming the MacOS version isn't entirely different from the Linux one):
[image]

xoxota99, 2019-07-02:
Wait, does that mean that if I'm printing (e.g.) 5 parts, each of them can have different layer heights? This doesn't really make sense to me, but maybe I'm mistaken. If I edit all the layers for one part to be 0.1mm, and edit all the layers for the part right next to it to be 0.2mm, how can they possibly print side-by-side? They'll have a different number of layers, won't they?

YuSanka, 2019-07-03:
Thanks for your report. We are working on new implementation for layers editing per object. It will be a part of upcoming release.

verner33ob, 2019-07-03:
@xoxota99: i guess it will just print the object with 0.2 layer height every second layer or only after the second 0.1 Layer is printed as well. Cuz then they are at the same layer.
The same happens e.g. with supports generated by PrusaSlicer. Only the highest layer at that point in time will be printed, so there wouldn't be any collision problems.

xoxota99, 2019-07-03:
@verner33ob okay, that works when the layer heights of *Two* objects are evenly divisible. What if I'm printing *twelve* objects, and have layer heights of 0.1, 0.13, 0.17, 0.19, 0.25, 0.3, etc...? The problem quickly becomes ridiculous.

verner33ob, 2019-07-04:
I never tried it and don't know how well Slic3r would handle it, but it shouldn't actually be a big problem.
Always the next layer that needs to be printed will be printed.
So smth like 0.1, 0.13, 0.17, 0.19, 0.2 (2nd 0.1 layer), 0.25, 0.26 (2nd 0.13 layer), etc.

supermerill, 2019-07-05:
slic3r sort layers by their height and print them in order. No problem.

bubnikv, 2019-08-01:
The layer editing will be back in the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-01 by bubnikv