prusa3d/PrusaSlicer #2588: Two layers with same orientation?

ochm, 2019-07-02:
### Version
2.0.0+win64

### Operating system type + version
Win8.1 64bit

### 3D printer brand / version + firmware version (if known)
i3 MK3

### Behavior
one layer - the bridge with some orientation of head movement
layer above the bridge - solid infill, the same orientation.
IMHO this is not correct.
In attached project on left front leg, layers 416,417, or at screenshots.

#### Project File (.3MF) where problem occurs
[image]
[image]
[Eiffel_Tower_Simplified_Base.zip](https://github.com/prusa3d/PrusaSlicer/files/3351608/Eiffel_Tower_Simplified_Base.zip)

supermerill, 2019-07-04:
The orientation of the bridge is computed to be optimal (but the program can be wrong).
The orientation of the solid infill depend only of the id of the layer (odd/even number).

So that case can happen, but it's really rare.
I don't know how to resolve that. Solutions can be worst than that.

DrLex0, 2019-07-17:
You can override the bridging angle in the Infill settings. If you don't want to override it across the entire model, you can add a modifier where needed.

closed 2026-09-29 by prusaslicer-bot[bot]