prusa3d/PrusaSlicer #2589: Printer Settings : Layer Height Settings cause crash

GWdd, 2019-07-02:
### Version
2.0.0  Winx64

### Operating system type + version
Windows 7 Pro x64

### 3D printer brand / version + firmware version (if known)
Prusa Mk3

### Behavior
Setting maximum layer height to 2.5 mm causes slicer to crash.

Using default settings, 
0.15 mm Quality, Prusament PLA, Prusa MK3 profiles
Load a part, add a part modifier: support enforcer and adjust to part
Select Supports for enforcers only (and accept the default print changes suggested)
Change Printer Settings: Layer height limits: maximum  to **2.5** mm
Slice

Slice will continue until supports are being rendered, and at then Plicer will crash.

[image]

#### Project File (.3MF) where problem occurs
[LayerHeightLimitCrash.zip](https://github.com/prusa3d/PrusaSlicer/files/3351655/LayerHeightLimitCrash.zip)

Plicer works as expected with 0.25 mm max layer height, crashes with 2.5 mm max layer height.

There is another issue posted where setting this value to 0 will also cause a crash.  Comments were "don't do that", which of course not acceptable.  Changing a setting to an illegal value should NOT cause a crash.  A exception warning should be thrown, or the bogus value not accepted when entered.

YuSanka, 2019-07-03:
Crash is caused by assert on negative flow value, because of wrong relations between layer's width and height.

	// Rectangle with semicircles at the ends. ~ h *(w - 0.215 h)
       height * (width - height * (1. - 0.25 * PI)) should be positive.
	   
Of course in this case application shouldn't be crashed and this bug will be fixed for next release. 
Till this moment note please the message in the tooltip. 

[image]

And there is one more small thing ;) Application name is "Prusa Slicer" not a "Plicer".

bubnikv, 2019-08-09:
Please test with the PrusaSlicer 2.1.0-alpha1. The issue should be fixed.

GWdd, 2019-08-09:
Changing layer max to 25 ...
[image]

Illicits this error message, which isn't really clear; but at lease now the app isn't crashing.
[image]

bubnikv, 2019-08-19:
thanks. Closing.

closed 2019-08-19 by bubnikv