prusa3d/PrusaSlicer #2590: PrusaSlicer embedded wrong version number in Gcode

pntucker2, 2019-07-02:
### Version
_Version of PrusaSlicer used goes here_2.0.0
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_OSX 10.14.5
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)MK3S mmu2s 3.7.2
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_ Gcode has wrong version # embedded and there is no "about" in the slicer menus
* _Steps needed to reproduce the problem_ read Gcode header info: look for "about" entry in slicer menu
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_ 
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2019-07-15:
I have followed your comprehensive list of steps to reproduce. PrusaSlicer version in the header of the gcode was correct, About dialog was accessible as always form Help->About PrusaSlicer. Please, provide more detailed info so we can actually reproduce the problem. Screenshots could help too. Thanks.

pntucker2, 2019-07-16:
When I checked again this morning I also got the correct version. I’ll dig through some of my  gcode and see if I can see when it Changed (and why)

Peter Tucker
  1 (604) 224 3928
 3675 west 29 Ave van. BC Canada. V6S 1T5

bubnikv, 2019-07-17:
Maybe that file was produced by some of the 1.42.0 alphas or betas?

pntucker2, 2019-07-18:
That could well be the case as I was using the betas

Peter Tucker
  1 (604) 224 3928
 3675 west 29 Ave van. BC Canada. V6S 1T5

bubnikv, 2019-07-19:
Could you close this issue then, please?

pntucker2, 2019-07-19:
I’d be happy to except I don’t see where to do it (this was my first use of github).

Peter Tucker
  1 (604) 224 3928
 3675 west 29 Ave van. BC Canada. V6S 1T5

closed 2019-07-19 by lukasmatena