prusa3d/PrusaSlicer #2591: Prusaslicer new features request

MR2C280, 2019-07-02:
New feature request.

Sorry if this isn't formatted correctly, this is my first post on GitHub.

There are a couple of minor feature in Repetier that would be nice to have in PrusaSlicer:

1) The ability to load a gcode file into the app so the user can review a slice done earlier. 
2) The ability to view, search and edit the gcode.
3) When a slicing parameter is changed, don't clear the preview and Slice Info immediately - wait until the user clicks "Slice".  This gives the user one last chance to see the previous slice and observe the effects of the change they just made.  To alert the user to the fact that the preview is out of date, you could change the background or font color. (If you really want to be nice, a pop-up that compares the previous Slice Info with the new one.)
4) At the top of the layer slider bar, show the total number of layers and total height.
5) Add the ability to set your default filament.  It currently always defaults to PLA and I usually use PETG so everytime I switch from printers (swithcing MMU on/off), I have to remember to change the filament.

iamgnat, 2019-07-02:
> Add the ability to set your default filament. It currently always defaults to PLA and I usually use PETG so everytime I switch from printers (swithcing MMU on/off), I have to remember to change the filament.

Really like this!

Similarly, if we could disable filaments that we won't ever use (similar to how we can select only our printer) that would make the filament menu much easier to work with.

supermerill, 2019-07-05:
> Sorry if this isn't formatted correctly, this is my first post on GitHub.

You should create one issue per request/bug. and so you can put a good name for the issue, like "request: add slice numbers on top the sliders"

imo (as a random guy that use slicers), 1) and 2) will be nice but there are the layer view that has been dropped and need to be re-done. The prusaslicer dev team won't have the time for big features like that before many months (years?).

5- there are a way to change it. It's not user-friendly, but you can open the PrusaResearch.ini that should be in your profile directory (like c:/users/myname/appdata/roaming/prusaslicer/vendor in windows) and change the values of the lines with "default_filament_profile"

iamgnat, 2019-07-05:
> 5- there is a way to change it. It's not user-friendly, but you can open the PrusaResearch.ini that should be in your profile directory (like c:/users/myname/appdata/roaming/prusaslicer/vendor in windows) and change the values of the lines with "default_filament_profile"

Not only is it not user friendly (removing unused profiles could be done the same way), you'll have to re-do it every time they update the configuration. There needs to be a setting that will stick across `PrusaResearch.ini` updates.

supermerill, 2019-07-06:
If you want to create your profile that isn't modify by the update of PrusaResearch.ini, you can create your own ini.

iamgnat, 2019-07-06:
The question isn't creating a new profile, that support is already there. The issue is changing which profile is the default (and I'd like to see that ability to disable/hide profiles).

supermerill, 2019-07-06:
You can't change anything in a vendor profile, but by editing the .ini

If you want to change the default of your profile that isn't the prusa, one, go to your setting folder, open your .ini (it in the profile directory, not the vendor one) and add/modify the default_filament_profile line.

MR2C280, 2019-07-07:
I still can't make it work.  If I make ANY change (even adding a space) to c:/users/_myname_/appdata/roaming/prusaslicer/vendor/PrusaResearch.ini then PrusaSlicer throws up an error upon opening and ALL setup settings are lost (rather defeats the whole point of a text based .ini file, in my opinion).  In fact, even making a copy and renaming it back causes PrusaSlicer to error out.  It's as if PrusaSlicer can detect if the .ini file has been tampered with in any way.

I tried placing a copy of PrusaResearch.ini up one level in c:/users/_myname_/appdata/roaming/prusaslicer and changing the default filament but it had no effect in PrusaSlicer.

Sorry, but not being a Windows programmer, I don't know which .ini file in which directory is used.  Where can I place a hand modified PrusaResearch.ini so that PrusaSlicer will use it?  

Am I wrong or is Prusa perhaps getting tired of novices playing with settings, getting bad prints and calling support so Prusa is trying to lock everything down to make it idiot proof?

rtyr, 2021-04-27:
2) implemented on master (without searching and editing, I don't see it as a task for a slicer). Issue renamed to reflect the last point.

closed 2026-09-29 by prusaslicer-bot[bot]