prusa3d/PrusaSlicer #2592: fine layers not allowed in the bottom of the print even when using rafts

Mirarkitty, 2019-07-03:
### Version
PrusaSlicer 2.0.0

Linux appimage version

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
* first layer can't be fine because of the contact with the heatbed
* rafts can be used to lift up the object from the heatbed
* **fine layers not allowed in** the bottom of the print **even when using rafts**
* expected to be able to set finer layers for the bottom of the object when the object is actually not touching the heatbed

#### Project File (.3MF) where problem occurs
[sphere_not_allowed_fine_layers_at_interface.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/3353913/sphere_not_allowed_fine_layers_at_interface.3mf.gz)

Note: supports hidden to make it clearer
[image]
[image]
[image]

supermerill, 2019-07-04:
because it use the "first layer height" and you can't really go low without scratching your build plate.

If you really really want it, use a raft.

Personally, i don't think the bottom of this object will be printable at low layer height, too much overhang & retraction will lead to a failure. You will have to use some sort of support.

Mirarkitty, 2019-07-04:
I *am* using a raft. That's what I'm complaining about. Look above, it says 4 layers raft. It's still doing that.

supermerill, 2019-07-05:
oh, sorry. i was tired.

I just remember that the raft can make the thing worst because the first layer on a support is a bridge, and the height have to be equal to the width on a bridge because you have nothign to push into.

To prevent the bridge, you have to use a soluble filament for the raft (Contact Z distance in support tab). Don't forget to set a filament change after it. You also should change the Interface pattern spacing to 0.

Mirarkitty, 2019-07-07:
Yeah, I know the raft is a bit daft but I was going to make another bug report for that... This is all about that I'm not even allowed to adjust the layer height. The red markings are "locked layer height".

[image]

bubnikv, 2019-08-01:
The first object layer over raft is always printed with bridging perimeters.

Mirarkitty, 2019-08-07:
Bridging layers can't have a layer height other than 0.02?

bubnikv, 2021-02-24:
Implemented with aee136cedd890ecb5072d9c9215a824fa3f365b1

The 1st obect layer over raft layer is newly printed with the normal flow, not bridging flow.

closed 2021-02-24 by bubnikv