prusa3d/PrusaSlicer #2594: Triple clicking fields in object manipulation does not select entire line

haarp, 2019-07-03:
PS 2.0.0 on Gentoo Linux, gtk+-3.24.8, wxGTK-3.0.4

A common UI feature is double or triple clicking text input boxes. A double click selections a section of text between separators, triple clicking selects the entire line.

This doesn't work in PS, e.g. in the the object manipulation boxes. The third click deselects the entry instead, making it very cumbersome to replace a value.

Cheers!

bubnikv, 2019-08-01:
Sorry, but I am afraid this is the implementation of wxWidgets on GTK.

Salamandar, 2020-04-07:
That 100% works for me. wxWidgets 3.0 AND 3.1, on Archlinux, PS 2.2.

haarp, 2020-04-07:
> That 100% works for me. wxWidgets 3.0 AND 3.1, on Archlinux, PS 2.2.

You're right, it seems to be working in 2.2.0. Cheers!

closed 2020-04-07 by haarp