prusa3d/PrusaSlicer #2595: reload from disk error after performing cut

OBSchofieldUK, 2019-07-03:
### Version
2.0.0+win64

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 Home Version 1809
OS build 17763.557

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK2S (Latest V3.2.3)

### Behavior
Problem occurs when a cut is performed on an STL, then right click and select reload from disk. a window pops up stating PrusaSlicer has encountered an error (see screenshot) 
"Unknown file format. Input file must have .stl, .obj, .amf(.xml) or .prusa extension."

* _Describe the problem_

* _Steps needed to reproduce the problem_
import an stl and perform cut anywhere (keep lower part only). 
Right-click and select "reload from disk"

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
Expects the STL to be restored to its original state
* _Actual Results_
Actually results in the error message as shown in the screenshot, part is removed from build plate after. 
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
no 
#### Project File (.3MF) where problem occurs
zip contains STL and .3MF file
[psCutProblem.zip](https://github.com/prusa3d/PrusaSlicer/files/3356305/psCutProblem.zip)

_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image]

vojtechkral, 2019-07-22:
Thanks. Yes, this is currently broken. Will look into a fix...

rtyr, 2021-02-02:
Fixed. Closing.