prusa3d/PrusaSlicer #2596: [Request] Making gyroid infill watertight

androandra, 2019-07-03:
### Version
Version 2.0.0+win64

### Operating system type + version
Windows 10 

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3s with MMU 2.0, running marlin firmware version 3.7.1-2266

### Behavior
The Gyroid infill is awesome, beautiful and stiff as well as strong. I really like it, so props for implementing that!

I recently learned that the gyroid is a doubly interpenetrating network, meaning that there are two seperate cavities like hinted at in the picture below.
[image]

This gave me a lot of inspiration to try and fill out the two cavities in the gyroid pattern with different dyes, glitter or other showy stuff. This would be especially awesome for prints with transparent filament! However I notice that the sliced infill is not actually watertight, as you can see on the screenshot here (layer height 0.2mm). It probably doesn't influence the structural integrity of the infill pattern too much, but it makes it impossible to have two separate cavities.
[image]

It gets much better when using a very small layer height, like 0.05mm, as you can see in the picture here. In this case it looks like it is completely closed. 
[image]
 
However it seems like overkill to print 0,05mm layer height just to get a bit more structural detail in very specific sections of the infill. This could be fixed manually by using variable layer height or modifier structures, but it's difficult to predict where the transition layers are gonna be when sliced.

My feature request is therefore to have a check mark option "Close gyroid infill watertight", which ensures that the layer height is decreased ONLY for the gyroid infill, and ONLY around the relevant transition layers.

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2019-08-01:
What benefits do you expect the watertight gyroid will bring? By the way, just extruding thinner layers will not work as you will still have extreme overhangs. You would have to over-extrude IMHO to fill in the gaps at the overhangs completely.

androandra, 2019-08-01:
I'm not sure I understand why it wouldn't work.
I would actually expect that printing extreme overhangs in e.g. domed structures is easier to print with a smaller layer height, since the new layer will have to cover less unsupported area. 

But yeah, over-extrusion would definitely help to close up any small gaps.

wavexx, 2019-08-21:
> What benefits do you expect the watertight gyroid will bring?

I'm also curious about this.

Structurally speaking, filling that gap completely won't bring any advantage with FDM as long as the layer height allows to support the underlying bridging, which is generally the case.

I agree that currently at some gyroid densities/layer heights/nozzle sizes the results are suboptimal. In these cases I manually increase the density.

androandra, 2019-08-22:
> > What benefits do you expect the watertight gyroid will bring?
>
> I'm also curious about this.

Well, not more than aesthetic pleasure. If it was watertight, then the two compartments would be sealed off from one another, which provides interesting options for filling them with various stuff. 

And I understand if it's not worth the effort to implement, but hey, it'll definitely never be implemented if no one posts the idea.

> Structurally speaking, filling that gap completely won't bring any advantage with FDM as long as the layer height allows to support the underlying bridging, which is generally the case.

I'm aware that I won't gain any structural strength or stability.

> I agree that currently at some gyroid densities/layer heights/nozzle sizes the results are suboptimal. In these cases I manually increase the density.

That works for me too, but because I can't predict where the gyroid pattern will be before slicing, I can't localize where to increase the density. The obvious solution is to increase density of the whole print, but it takes sooooo much longer!

wavexx, 2019-08-22:
> Well, not more than aesthetic pleasure.

I do completely agree with this :)

[image]

This solid was constructed manually and printed "just for looks". I manually adjusted the layer height at the overhangs using the built-in variable-height feature. It did work pretty well. There are still gaps, but that was mostly my fault as I printed this completely hollow. Solid infill would have closed the volume.

> That works for me too, but because I can't predict where the gyroid pattern will be before slicing, I can't localize where to increase the density. The obvious solution is to increase density of the whole print, but it takes sooooo much longer!

Thinking out loud, gyroid infill *could* hint the built-in variable height map in order to reduce the gaps, but infill by itself doesn't have the ability to trigger extra rectilinear areas to close the gaps you see above.

aleksclark, 2021-03-12:
I have a structural use case that would benefit from making the gyroid surface continuous regardless of layer height: I'm printing airplane wings. Gyroid infill seems to offer a higher degree of rigidity/weight than other infills, but at low infill % (1-2%) there are a lot of gaps. It seems like some sort of scaling could be applied to the spacing, so that the gyroid looks 'stretched' but there is enough overlap between the layers to allow even further reduction of the infill %.

pmarks-net, 2024-03-03:
[image]

Here's a half-baked idea: Normally gyroid transitions directly from the red pattern ⩆ to the blue pattern ⫗.  What if you add this hypothetical green pattern that bridges the gap using Z/N-shaped squiggles instead?

In order to transition from green back to blue, the green layer would also need to include the missing blue segments (a second pass with retractions.) Or maybe it's possible to ignore blue and just continue on with the green pattern?  I have no idea what that would do...

pmarks-net, 2025-08-13:
I think the problem is that the gyroid shape contains overhangs beyond 45 degrees, so it fundamentally cannot be produced on an FDM printer without supports.  This problem only becomes relevant when you print with a large cell size, larger than people would typically use for infill.

To make a large gyroid with watertight channels, I think you'd have to model the gyroid shape as an STL with appropriate thickness.  At some scale you'd need to start including internal supports.

However, Slic3r-style 3D Honeycomb infill (https://github.com/prusa3d/PrusaSlicer/pull/6434, never merged into PrusaSlicer) is structurally very similar to gyroid, but with 45 degree slopes.  It still has holes, but the holes are square, and it wouldn't be that difficult to cover them with a 2-3 layer bridged roof.

In other words, Slic3r-style 3D honeycomb + square roofs would probably make a good heat exchanger.

gringer, 2026-09-07:
I've been working on an OrcaSlicer update to add tops to my truncated octahedron infill (3D Honeycomb), which will allow a similar watertight function to what you're proposing here. This approach works best with multi-line infill because that allows rotated top layers and a bit of extrusion squishing (reducing the chance of gaps), but I could port the single-layer top across to PrusaSlicer if there were interest in it.

This is all conditional on the 3D Honeycomb being updated to have the wall fix that went into Slic3r a few months after PrusaSlicer was forked from it, as in this PR:

https://github.com/prusa3d/PrusaSlicer/pull/6434

For more details on the top-filling option, see here:

https://github.com/OrcaSlicer/OrcaSlicer/pull/12541

Jan-Soustruznik, 2026-10-01:
Hello, @gringer, thank you for update, According to new [policy](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I closing this feature request and converting it into a Discussion thread to hear what the community thinks about this feature. Thank you, and have a nice day.