prusa3d/PrusaSlicer #2597: on 3D tab, not showing full model

KB7GP, 2019-07-04:
### 1.41.3+win64

_Use `About-
[image]

### Windows 10 64 bit

### PRUSA i3 mk3

### 3d model is not represented on the print base (see the image) upon loading the STL file.
Not a new feature request

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

supermerill, 2019-07-04:
please

Upload a PrusaSlicer Project File (.3MF) (Plater -> Export plate as 3MF for Slic3r PE 1.41.2 and older, File -> Save / Save Project for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)
Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)

KB7GP, 2019-07-04:
Ok will try... However, Since update a few months ago, All STL files I processed, only showed the body contact point on the plate representation (The green contact area). Before upgrade, the whole model was represented on the slicer plate screen.

The model does print normally and in viewing the Layer tab, I can move through the entire model... no change there.
Dave

From: Merill 
Sent: Thursday, July 4, 2019 6:23 AM
To: prusa3d/PrusaSlicer 
Cc: KB7GP ; Author 
Subject: Re: [prusa3d/PrusaSlicer] on 3D tab, not showing full model (#2597)

please

Upload a PrusaSlicer Project File (.3MF) (Plater -> Export plate as 3MF for Slic3r PE 1.41.2 and older, File -> Save / Save Project for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)
Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)

KB7GP, 2019-07-04:
Not sure if I uploaded to Github. I have attached the file here.
DaveR KB7GP

From: Merill 
Sent: Thursday, July 4, 2019 6:23 AM
To: prusa3d/PrusaSlicer 
Cc: KB7GP ; Author 
Subject: Re: [prusa3d/PrusaSlicer] on 3D tab, not showing full model (#2597)

please

Upload a PrusaSlicer Project File (.3MF) (Plater -> Export plate as 3MF for Slic3r PE 1.41.2 and older, File -> Save / Save Project for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)
Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)

KB7GP, 2019-07-04:
Ok... attached to email.

supermerill, 2019-07-05:
you should drag-drop the zip here, in the text area.

rtyr, 2021-01-28:
No response. Closing.

closed 2021-01-28 by rtyr