prusa3d/PrusaSlicer #2598: improper 1st layer support material overlap object

ohthetrees, 2019-07-04:
### Version
PrusaSlicer-2.0.0+-201905201652

### Operating system type + version
MacOS 10.14.3

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

I'm using multiple materials the old fashioned way: on a single extruder MK3S I set up multiple extruders in Prusaslicer and issue a M600 Gcode command on tool change. Works great. I use it to print inlays into the first couple of layers of an object. This technique relies on multiple stls that fit together like puzzle piece, then I assign different "extruder" to each stl. 

Problem is that if I have support material, these weird little spurious pieces of support material get put down on the first layer here and there. When printing one stl, they tend to be harmless, but aesthetically annoying. But in my situation, these little random squiggles of support material, which don't support anything or serve any function, ruin my inlay effect.

As far as I can tell, two separate problems are combining to make me miserable.

A) For some reason Prusaslicer will make little random bits of 1st layer support material, even if they don't support anything.

B) Prusaslicer doesn't respect another object when making support material. I found #316 which seems to be a similar bug, but guys, it is from 2017, and still no resolution. 

I'm filing this as a new bug, because #316 alone wouldn't cause this problem for me, my problem A is also necessary. 

Please take my complaint in good spirit. I really like Prusa pinters and Prusaslicer. Thanks for all the hard work!

[image]
[image]
[image]

#### Project File (.3MF) where problem occurs
This is not a "real" file I use, the model has some problems, but it demonstrates the issue I'm talking about.
[Demo support problem.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3358232/Demo.support.problem.3mf.zip)

supermerill, 2019-07-04:
As a short-term fix for this "bug", you can use support enforcer to exclude the area where you don't want them (the 'bits').

ohthetrees, 2019-07-04:
Unfortunately, support blockers don't block the "bits". I tried on both the stl that needs the support (the case) and the stl that is interfered with (the "e" stl).

neophyl, 2019-07-05:
Merril said support ENFORCERS not blockers.  There is a world of difference.  I have attached a modified version of your 3mf with a single support enforcer as an example for you to work with.  

Essentially you turn off automate support and then put enforcers for the area where you need it.  You can play around with the size of the enforcers to fine tune the area.  They are much better than blockers for some reason.  Since the betas and now 2.0 allowing much easier placement of modifiers Ive stopped using blockers completely and just use enforcers.  

[image]

As you can see no squiggly bits.  

[Demo support problem_ENFORCERS.zip](https://github.com/prusa3d/PrusaSlicer/files/3361611/Demo.support.problem_ENFORCERS.zip)

ohthetrees, 2019-07-11:
Thanks @neophyl, this workaround worked for me. Of course, I think there is an underlying bug that should be addressed, but at least I can get my desired result.

closed 2026-09-29 by prusaslicer-bot[bot]