prusa3d/PrusaSlicer #2599: issues removing supports

d2crafty, 2019-07-04:
### Version
prus ver. 2.0.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
windows 8.1

### 3D printer brand / version + firmware version (if known)
flashforge creator pro

### Behavior
prints come out fine, however the supports seem to weld to the part.  ive tried various settings to e.g. the XY separation at 75% however the problem still remains. .

_Is this a new feature request?_

neophyl, 2019-07-05:
Have changed the Contact Z distance ?  by default its set to 0.1 but I find that to do exactly what you describe ie stick them on really tight.  Its located in Print Settings > Support Material same  as the XY seperation..   I change mine to 0.2 or sometime 0.25 and find them MUCH easier to remove.

d2crafty, 2019-07-05:
I'll give that a shot to ight and see the results.. I resorted to just adding more blockers to reduce the amount of supports, however I feel the supports shouldn't be that difficult to remove

bubnikv, 2021-02-24:
Stalled issue, I suppose changing contact Z distance helped.

closed 2021-02-24 by bubnikv