prusa3d/PrusaSlicer #2600: Support Blockers - not working

adamleeschwartz, 2019-07-04:
### Version
2.00

### Operating system type + version
Windows10

### 3D printer brand / version + firmware version (if known)
MK3

### Behavior
Blockers do nothing to change the support structures on this model. We would like to unsupport certain areas of the model in order to retain detailed columns in those areas. 

_Is this a new feature request?_ 

nope

#### Project File (.3MF) where problem occurs

[19 06 25 _70KING_1-1000 Assembled (1)_building.zip](https://github.com/prusa3d/PrusaSlicer/files/3360140/19.06.25._70KING_1-1000.Assembled.1._building.zip)

[image]

[image]

neophyl, 2019-07-05:
Support blockers in my experience are blunt tools rather than scalpels.  By that I mean they dont work in the precisely defined area that the blocker covers.  I think itd down to the fact that supports to work cover alot more surface area than the support itself to give it a base to stand on.  That means in practice that any blocker needs to be bigger than the area you want to block.

[19 06 25 _70KING_1-1000 Assembled (1)_building_extra_blocker.zip](https://github.com/prusa3d/PrusaSlicer/files/3361557/19.06.25._70KING_1-1000.Assembled.1._building_extra_blocker.zip)

Iv'e added an extra blocker to your 3mf in the upper left to illustrate what I mean.  Because PSlicer is trying to support the edge of the overhang still it fills in the area around it.

btw with your print settings for 0.1mm separation of the support on PLA Iv'e found it very hard to remove them, changing it to 0.2 (which is the default for detachable) makes them much easier to remove, 

 Play around with different blocker sizes to see what is needed.  Thankfully that is much easier to do in 2.0.

adamleeschwartz, 2019-07-05:
Yes I am aware of all of this, but thank you for your response. Perhaps I should have made this a request for a new feature. Thanks anyways.

closed 2019-07-05 by adamleeschwartz