prusa3d/PrusaSlicer #2601: Feature Request

FilamentFrenzy, 2019-07-04:
### Behaviour
* _Feature request : optimise travel moves when printing multiple objects or a single object with multiple travel islands by starting a new layer on the last printed object per layer instead of moving to the first one!

practical-printing, 2019-07-04:
I agree this would be handy, at least as an optional feature.  Not only does it reduce travel between objects speeding up print time, but it also minimizes retraction related artifacts (such as stringing) by reducing cross-plane travel to every other layer.

JohnOCFII, 2019-07-05:
Hmm...  Pretty sure I've seen PrusaSlicer do this on my prints.

Mirarkitty, 2019-07-05:
I thought it did this already, but if it doesn't it should!

supermerill, 2019-07-05:
please rename your issue with something meaningful like "request: invert order of islands on odd layers"

@ JohnOCFII it can with mmu mode. But it works by extruder, not islands.

DrLex0, 2019-07-17:
This is basically the same issue as #2614 and my same comment applies here: for small parts this may not be a good idea if consistent print quality is important, because cooling of the material will become inconsistent. If there are 4 objects, or one object with 4 pillars, then two of them will have pairs of layers that are printed immediately after each other while the other two will get more cooling time in between layers. This reverted ordering planning would be nice to get faster prints, but I want to keep the option to always print in the same order as well.

Mirarkitty, 2019-07-18:
Yes, I think having this as an option like _[ ] slice all objects as one object too optimize travel_ would be good.

haarp, 2019-07-21:
I have an example on #2614 which shows what this is about. But I agree with @DrLex0. It might or might not lead to inconsistent looks. Having it as an option would be ideal.

dartrax, 2021-09-20:
fyi: Simplify3D has this, called „Print Islands sequentially without Optimation“. It is on by default. I disable it only for small parts.
IdeaMaker also has it, to disable select „Print Parts in Same order for each layer.“

JonBerlin, 2021-12-30:
Please add this feature! This would help so much with printing with big nozzles where oozing is uncontrollable.

Raph22, 2022-05-31:
Yes print islands sequentially in Prusa Slicer would be super cool !

closed 2022-09-13 by FilamentFrenzy