prusa3d/PrusaSlicer #2602: [Feature Request] STL preview in Windows

bivaterl, 2019-07-05:
2.0 Version

Win 10 Pro

Prusa Mk3, firmware 3.7.2

NEW FEATURE REQUEST

I have been trying to find a way to get the STL files to show as mini models. This works natively on mac with preview, but it would be nice to see the model without having to open in slicer. can prusaslicer be configured to save the med/large icons as models instead of as the PS logo?

Jebtrix, 2019-07-05:
STL Thumbnail
https://marlin3dprintertool.se/stl-thumbnail/

bivaterl, 2019-07-05:
Thanks, Jebtrix, I have this installed and I can open a viewer to see these (very quickly, I might add), but it doesn't save the image as the preview. I will work on the process outlined in which I delete any and all cached images and then replace with the marlin 3d tools images, to see if I can get this to work. In the meantime, I'd love to see this built into a future version of PS... It was one of those things I didn't realize I missed moving from my old Mac to the new PC.

Jebtrix, 2019-07-05:
I'm not sure what you mean by save the image as a preview. When the STL thumbnail is created after viewing it in File Explorer/Windows Explorer it is cached in:
 C:\Users\<username>\AppData\Local\Microsoft\Windows\Explorer\thumbcache_<pixelsize>.db

The preview image is indeed "saved".  Depending on how far you zoom in (Small Icons, Medium Icons, etc) the STL thumbnail is re-rendered (just once) at that size. Because its an extension to Windows builtin thumbnails you are still bound by OS thumbnail rules. This includes if you have turned off thumbnail caching, or if you have remote/network drive caching enabled or disabled. Easy way to clean cache is just launch Disk Cleanup utility and just check Thumbnails in list.

rtyr, 2021-02-02:
Implemented for 3MFs, I believe we can close this.

closed 2021-02-02 by rtyr