prusa3d/PrusaSlicer #2603: Scale and UI out of sync

GWdd, 2019-07-05:
### Version
2.0.0win64

### Operating system type + version
Windows 7 Pro x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
Due to "autoslice" happening even though "background processing" is disabled, the UI and the the settings get out of sync. In this example, I modifed scale from 140 back to 100% and Plicer is still slicing at 140%, even though the UI clearly shows 100% is what I want. Switching between PART and SLICE views has no effect. Changing BRIM on - off - on to force a reslice (and it does autoslice again) has no effect on scale: scale remains 140% despite the 100% setting.

Verify by seeing Z part size as 148 mm, yet print size is 207 mm.
[image]

It would be nice to have Plicer not autoslice; and to actually obey UI settings.

#### Project File (.3MF) where problem occurs
[UI Sync.zip](https://github.com/prusa3d/PrusaSlicer/files/3363841/UI.Sync.zip)
I've included the original part since the 3mf resizes it to the currect scale settings.

bubnikv, 2019-08-01:
Sorry, I cannot make a sense of it. If you have "autoslice" disabled, then it will not autoslice. It will continue to slice until you change a value. If you change a scale to 140%, the value is first accepted when you press enter or leave the field. If you do that and "autoslice" is disabled, the background processing will stop.

GWdd, 2019-08-01:
The issue is Autoslice IS disabled (background operations) - yet the program keeps autoslicing every time I change a value. 

Perhaps there is another "autoslice" control I have not found?

bubnikv, 2019-08-01:
>  yet the program keeps autoslicing every time I change a value.

What do you mean with that? Why do you think that the application autoslices something?

GWdd, 2019-08-01:
I make a change to a value - and the program begins slicing, and after it completes the slice it begins doing the post process and eventually shows the final multicolored colored display with all of the features identified.  And all I did was change scale.   

I believe it might be related to the autoslicing done when you change from the part view to the slice view.   So that when you are viewing the layers, it presumes you want it sliced with every change.

GWdd, 2019-08-01:
Then there's this example: I change the part view scale from 100 to 50, with all scales linked (pad-lock) and yet the part scale doesn't actually change and the two other "linked: dimensions also don't chenge.
[image]

GWdd, 2019-08-01:
I think I may have discovered what the actual issue is, that NOTHING is selected: even though the part moai-new.stl is highlighted and looks selected, it isn't blue and is NOT selected. Thus, the scale controls shown are disconnected from everything.  Which begs the question: why are they being shown at all?. 

[image]

GWdd, 2019-08-01:
And if I make that size change when in preview, the app reslices. Arrgh!

Seriously, Voj, great tool, but it has some UI quirks that are pretty annoying.

YuSanka, 2019-08-09:
Re-test, please, this issue with upcoming PrusaSlicer 2.1.0-alpha1. This bug looks like fixed.

GWdd, 2019-08-09:
It is really annoying that adding the new version erases all of my existing customer profiles.

GWdd, 2019-08-09:
Changing scale in plater view now seems to update all scales properly.

Changing scale in slice view now changes all scales as expected, but exposes another inconsistency.
When in plater view and clicking slice view, the application begins a slice operation (regardless of background setting).   So when in slice view, and changing scale, I'd expect the autoslice to begin.

I am very happy it doesn't, that the preview simply is cleared and I am given the SLICE button.  Because I may want to change more than scale before reslicing.   But it is inconsistent with selecting the window.

Perhaps selecting the slice window should NOT start a slice when background operation is disabled.  Let the user chose when to start slices at all times (this was how Slic3r operated).

YuSanka, 2019-08-14:
> It is really annoying that adding the new version erases all of my existing custom profiles. 
....

Please, don't mix different issue in one place. For new/another bug/FR just create a new issue.

YuSanka, 2019-08-14:
> Changing scale in plater view now seems to update all scales properly, and view is updated correctly.

Thanks for re-testing with new release. 

> Perhaps selecting the slice window should NOT start a slice when background operation is disabled. Let the user chose when to start slices at all times (this was how Slic3r operated).

Every customer has own suggestions ;) But... expected behavior is:
1. If background process is ON, each change forces re-slicing.
2. If background process is OFF, for new slicing you should do one of two : push button "(Re)slice now" OR switch from "3D editor view" to "Preview" scene mode.
Such behavior was approved by Prusa owners.

bubnikv, 2020-11-24:
Tim, is the original issue resolved for you? If so, would you please close this issue? Thanks.
By the way, my name is Vojtech or Vojta, not Voj.

GWdd, 2020-11-24:
> Tim, is the original issue resolved for you? If so, would you please close this issue? Thanks.
> By the way, my name is Vojtech or Vojta, not Voj.

Apologies for the name shortcut ... my name is a lot longer than Tim, but in American English it is customary to shorten names - Robert to Bob, William to Bill, Richard to Dick, etc.

Later ...

closed 2020-11-24 by GWdd

bubnikv, 2020-11-25:
> Apologies for the name shortcut

no offense taken

> my name is a lot longer than Tim, but in American English it is customary
to shorten names - Robert to Bob, William to Bill, Richard to Dick, etc.

We do shorten the names as well in Czech language, but I suppose the
grammar rules are more involved than in English. Some names may shorten the
way you did it, some not.