prusa3d/PrusaSlicer #2604: Avoid crossing perimeters doesn't do that

BirkBinnard, 2019-07-05:
### Version
2.0.0+win64

### Behavior
[image]

_Is this a new feature request?_

Not really. I know the crossing perimeter option was discouraged from use in earlier versions. I guess it is still a WIP.

aphex3k, 2019-11-07:
I think "avoid crossing perimeters" has a serious problem with cylinders.

bubnikv, 2020-12-11:
> I think "avoid crossing perimeters" has a serious problem with cylinders.

I would say the "avoid crossing perimeters" feature had serious problems NOT ONLY with cylinders :-)

"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

No model was provided, so we just hope that our reworked algorithm works better with this model.

closed 2020-12-11 by bubnikv