prusa3d/PrusaSlicer #2605: Feature request: Adjustable Brim Height

Pun-e, 2019-07-06:
### Behavior
When I go to make my brim higher, I can see a height option for the skirt, but nothing for a brim.

### Expected Behaviour
That there is a "Brim height" dialogue identical to the "Skirt height" dialogue.

_Is this a new feature request?_
Yes.

This has been an [ongoing request for 7 years](https://github.com/slic3r/Slic3r/issues/641) that the main branch of Slic3r have determined as a "waste of time".

This option would be a major, if not essential, boost for:
- Materials that have weak inter-extrusion bonding but strong warping, and
- Prints that occupy so much of the bed that there is no feasible way to increase the brim's width.

I made a bit of a stink over there about this is why Prusa-Edition is such a big deal. I'm an engineering undergrad with _years_ of printing under my belt. I've spent countless hours perfecting PC (Poly-Carbonate) printing on my Prusa mk2 and this option would've made all the difference.

A great example would be when I had to reprint all of my x-carriage in PC because it was starting to be damaged by the continual use under such high heat. Prints would continuously fail when the PC would pull away from the brim that was still stuck to the bed. In the end I had to settle for glue-stick, printing each piece one at a time, and using a massive skirt (both in width and height) to achieve success.

This feature would both be practical, and a statement about the level of functionality and user-understanding that separates PE from its competitors, so I hope it's something that can be implemented in the future.

zapta, 2019-07-06:
Pun-e, thanks for posting this feature requests. Control over brim and support base on bed is a common feature on the other slicers I have been used and is at least as useful as the skirt height control in slic3r.  

As per your Prusa-Edition reference, is this a kind of feature that Prusa may implement independently in their own fork? If so, do you know if it was raised or discussed there?

lordofhyphens, 2019-07-07:
@bubnikv Just as a FYI, I did a simple hack to the C++ CLI. I have no interest in going through the work to bring it to the GUI (because that would also affect preview, etc, and there's enough going on that I don't want to go through the effort for a feature that I don't believe will solve the actual problem at hand).

I have had zero reports of anyone actually trying to use it and I don't see this use case and have no time to do the experimentation.

https://github.com/slic3r/Slic3r/tree/feature-641

lordofhyphens, 2019-07-07:
@bubnikv https://github.com/slic3r/Slic3r/pull/4836 for a quick pointer to the code if your team wants to spend time on it, but that branch should give enough baseline for people trying to show it helps.

Pun-e, 2019-07-11:
A perfect demo of print that slic3r cannot currently handle:
[image]
A - Where part separated from brim
B - The entire 6mm brim that was stuck so hard to the PEI I damaged it during removal

Things already done to combat this behaviour:
- Brim
- Enclosure (close to 40 C)
- Bumped extrusion to 290 C
- Bed at 100 C
- Larger first layer
- Lowered the first level height significantly
- Gluestick (not on this particular print)

As I said originally, at this point a simple multi-layer brim could prove the difference. I have been provided with a [platform to test multi-layer brims](https://github.com/slic3r/Slic3r/pull/4836), and am currently printing the exact same part but the only difference will be a 4-layer brim.

If this part doesn't fail then I hope that this will be considered again.

bubnikv, 2019-08-01:
@lordofhyphens thanks for heads up.

There are the following issues when implementing a feature that is useful to a tiny fraction of the use base:

The day has 24 hours and there are more pressing issues in the slicer to solve.
Making the code more complex costs more time and money in the long term to maintain.
More parameters means user confusion.

For now, I can recommend you to add cylinders at the corners as the "mickey ears" and set their height tho the required number of layers. You can add the cylinders directly in PrusaSlicer in the expert mode.

Falk3r, 2019-11-29:
I, too, would appreciate this option.

Camelfoe, 2020-01-18:
Could someone explain what possible use there is for a multilayer skirt?  I think a multilayer brim would be much more useful, and also help when printing tall, skinny prints.

Falk3r, 2020-01-18:
> Could someone explain what possible use there is for a multilayer skirt? I think a multilayer brim would be much more useful, and also help when printing tall, skinny prints.

Poor man's thermal enclosure?

lordofhyphens, 2020-01-21:
Tall skirt has a few uses. One, it acts as a thermal enclosure.
Two, it is a cheap place to do nozzle purges.

mateuszbrzezinski, 2020-07-19:
I would add another reason to customize brims, if the brim has only 1 layer, it is sometime hard to peel it from the bed, if it would have at lest 2 layers, then it would be much easier to take it from the bed

Cologan, 2021-02-07:
yes, this please. I have had to go into CAD and add brims manually for more difficult prints. OR i print rafts just to emulate a brim (rafts stick out a bit + normal brim around raft, but it still has issues sometimes with warping). Thicker brim built into slicer would be major QoL for me (i need thick brims quite often).

bcarteruk, 2021-02-15:
I have the same problem - a single layer brim is not enough to hold the corners down on PLA+ filament - so I have resorted to adding manual multi layer brims in the CAD file.

SovereignGFC, 2021-04-30:
> I have the same problem - a single layer brim is not enough to hold the corners down on PLA+ filament - so I have resorted to adding manual multi layer brims in the CAD file.

I had to do this too, though I simply added a brim on the inside since it was a hollow object (four sides of a box too tall to print in one go even on a Prusa).  It's usually sharp angles (90 or less--corners, points) that have this problem.  Maybe add a "reinforce corners" option.  I don't know how feasible this is, but this is the idea:

- Allow the user to place resizeable manual "dots" or "squares" where the brim is thicker.
- Detect 90-or-lower angles and automatically thicken the brim for ~2cm along each edge.

LightningShark25, 2021-06-07:
Respect there are other higher priority jobs in the queue but would appreciate this also.
Having trouble with tall ABS prints breaking away from the brim itself.

My manual workaround is adding the brim in software, or applying superglue while printing, but a feature would be nice

Cheers

LukasM82, 2021-11-04:
Just came across this post looking to add some multi layer brims for materials that warp easily such as PC.

Would be fantastic to have this option in PrusaSlicer!

adamfilip, 2021-12-03:
+1 need this feature, id rather not have to manually create Mouse ears in the STL.

asdasd-123, 2021-12-28:
This feature is the last one stopping me from making the complete move over to PrusaSlicer. I print a lot of warping materials and a thick brim is the quickest/simplest/cheapest option to stop them warping at all. Would love to see it added.

sietzehouwink, 2022-02-13:
I'm printing a large object that still warps after having tried many fixes. A multi-layer brim would likely solve this issue. I understand there are (always) more pressing issues, but this feature should not be too time consuming to implement (naively).

Tianziegler, 2022-08-15:
Back in the day I used simplify3d and used multilayer brim. I miss the function in prusaslicer a lot.

mpaw, 2022-12-11:
@Pun-e and whoever else might find this useful, I have a workaround for this:

1. Create a box (or cylinder) that is larger than your part by the distance you want the brim to go in X/Y.

[image]

2. Create a copy of your part, merge it with the box, and change the type to `Negative Volume`

[image]

3. Center the box on the negative volume part and slice it. It should look like this

[image]

4. Select your whole merged group and use the `Cut` tool. Change the `Z` value to the height you want your brim to be and select only `Keep lower part` and click `Perform cut`

[image]

5. Right click on your `Merged` group and click `Layers and Perimeters`. Under `Perimeters` change it to `100` (or some other large value that will cause the merged group to be made up of only perimeters). Also make sure that your elephant foot compensation is turned off.

6. Center your merged group on your original part and slice. You should end up with something like this:

[image]

[image]

In the example above you can see that the center of the object also has the brim. If for instance you want to emulate the `Brim Type: Outer brim only` functionality you can add an additional negative volume to your merged group that covers that internal space. 

[image]

Obviously this isn't as convenient as a setting in the UI but it's not too difficult to do and works.

1tein, 2023-04-04:
Any progress on the multi layer brim function?
It would be really handy!
Thanks a lot!

Insane88, 2023-04-14:
same issue, particular need for lithophane photos, small bed footprint, need sturdy brim

AustinTylerDean, 2023-06-25:
Also +1 for this! I think it could also help with certain materials like ASA that warp. Mouse ears are NOT easy in Prusaslicer compared to other slicers (Cura cough cough) and I even had some mouse ears refuse to render until I reloaded the slicer! 

With an extra couple layers of brim, we could get the strength necessary to overcome slight warping, and if need be, hopefully someone will add the ease of mouse ears to PrusaSlicer in the future. At that point we have a couple tools for the job.

GUSTAVOANDRECUNHA, 2023-10-05:
Today I am printing Nylon and I see this function as something that can really solve my problem, but there is no option to add more denim layers in Prusa04/10/2023 please add this function, I will have to model this tab in CAD

annevanleyden, 2023-11-04:
I have found the multilayer brim a lifesaver for printing flexible and warp-prone materials like PA. The single layer brim suffers from the peel-away effect. It just comes loose slowly, mm by mm, during the print due to the constant force of the object warping. Because of this, making the brim wider doesn't help. 

Two things are important: 
1. The brim separation gap can and should be slightly larger for the higher layers. One layer of brim/object attachment is enough in my experience. This will make the multilayer brim as easier to remove than a single layer brim. Having that small gap on all layers will make it hard to remove and leave scars on the outer wall.
2. The brim should have a triangular cross-shape to prevent it from warping itself. It should get gradually thinner towards the outside. This is a crossection of the brims I have been adding to my designs by hand.
[image]

This significantly and reliably reduces warping. But it is time consuming to add these manual brims to each design. Especcially for complex shapes. Please (re)consider this feature.

FoldZero, 2024-02-11:
Just to add, PETG at .10 layer height is a murder to get off a pei plate. 
I was looking for the 2, (3, 4...) layer height setting for the beim tonight, whice has led me here.

surprised really its not already art of the options, as 
[image]
is just above this family of settings.

Please reconsider

WalterWolf49, 2025-07-23:
+1
With smaller layer heights brims and especially mouse ears are extremely difficult to remove from the build plate, as they are simply too thin. The option to increase the height is absolutely required!

zapta, 2026-09-02:
This feature request is just relevant today as it was in the day it was created. Please keep it open and address it.

annevanleyden, 2026-09-03:
@Kellegram 

> if someone has warping issues this is not a solution.

I have created these thick brims in my CAD program for the sake of reducing warping and can from first-hand experience tell that they work better than single-layer brims.

Kellegram, 2026-09-15:
> [@Kellegram](https://github.com/Kellegram)
> 
> > if someone has warping issues this is not a solution.
> 
> I have created these thick brims in my CAD program for the sake of reducing warping and can from first-hand experience tell that they work better than single-layer brims.

I think you misunderstood. What I mean is, that's a band-aid, one should ideally address what causes the filament to warp excessively if possible. It's not a "solution", but it is a band-aid that will work yes, I'd be surprised if it didn't.

jera-sea, 2026-09-15:
> I think you misunderstood. What I mean is, that's a band-aid, one should ideally address what causes the filament to warp excessively if possible. It's not a "solution", but it is a band-aid that [...]

That doesn't make sense. You can't just "address the cause of filament warping".
For any given print, a person is sitting in front of a printer of a specific model with some filament loaded. At that point there are no options to reduce warping further. They have the model they need to print and the warping dynamics are already decided. The only lever at this point is a thicker brim.
You can't just say "address the root cause of warping" that's simply not possible for 99% of people. Even with all other parameters perfectly optimized, some models will cause ABS, ASA, PLA+ etc. to warp and pull free from the typical print bed.

I think the reason this feature request is sitting ignored for so long is that the slicer developers are using high end printers and quality brand name filament, so they don't have the problems that the average user has (cost constraints) and they think it's just noobs failing to choose the right print settings.

Kellegram, 2026-09-15:
> That doesn't make sense. You can't just "address the cause of filament warping". For any given print, a person is sitting in front of a printer of a specific model with some filament loaded. At that [...]

"High end filament", as in purer than the blends we get, actually warps more. Really aside from PLA, PETG and TPU, anything else you will print is highly suboptimal anyway, abs/asa needs like 80-90C chamber to even get full material properties, that includes all the blends. A tall brim is a band-aid to using more heat, heatsoaking, using various preventative settings and design and so on. I'd recommend just using a strong adhesive like NPA or one of the DIY alternatives like goo by frank, or use magigoo. That way you don't have to wait for tall brims to band-aid your warping problems. The material will just warp the plate instead if it's warping strongly. NPA/DIY solutions can be so strong that you can print peek/PC open-air (badly ofc, the material properties will be trash, but the fact it works at all speaks a lot about these adhesives). 

If you have PLA warping off of the bed, that's absolutely a settings issue lol, or you have a draft, the pla is wet (wet filaments can warp more too) and so on.

jera-sea, 2026-09-16:
nah @Kellegram You still missing the point.
Yes you CAN solve warping in other ways (e.g. adhesives). But all other fixes require sourcing and buying more materials or tools, trial and error or printer modifications.
A thicker brim is a fix for anyone with a printer with zero additional effort.

You can say, "just use the right build plate for your filament type". But what if I only have one build plate? You saying I must go source and buy more stuff when I could just change one slicer parameter and the print will work.

zapta, 2026-09-16:
The question if brim is useful for X or not is not relevant here. The Prusa Slicer does support brim and this issue is for making it easier to remove from the build plate..

WalterWolf49, 2026-09-30:
> The question if brim is useful for X or not is not relevant here. The Prusa Slicer does support brim and this issue is for making it easier to remove from the build plate..

This brings it to a point. Arguing about brims does not help. Brims and mouse ears help, thats a fact and the reason Prusa Slices supports brims and mouse ears. Multi layers help in actually making those features viable with thin layer lines (as in being able to remove brims and mouse ears).