prusa3d/PrusaSlicer #2606: UI Request

prowlmedia, 2019-07-06:
UI Issues.

The Plater / Slice view Should be at the top - Make no sense have It bottom left. 

3D - Layers  ( Gap )  -  Print  Settings - Filament Settings - Printer Settings 

Also Slice now and Export GCode Bottom right is odd.  Again these should be at the top.

On a big screen I am constantly flying around from corner to corner - There is a reason that Buttons tend to be grouped at the top!

bubnikv, 2019-08-01:
Thanks for your suggestion, but we will not change it.

closed 2019-08-01 by bubnikv

prowlmedia, 2019-08-01:
Clearly not bothered with a usable UI then.

prowlmedia, 2019-08-01:
Actually. If those icons could be customised and moved around then you’ll have something great. I know it’s the base code you are working with but something to think about.

lukasmatena, 2019-08-02:
> Clearly not bothered with a usable UI then. ;)

No, we just understand that everyone has his own opinion about how the UI should look and it's not possible to make everyone happy. If we moved it, someone else would come and write that the placement is crap and we clearly don't bother about having a usable UI. Which is why it is really wisest to keep it the way it was for years. I agree with @bubnikv. Sorry.

> If those icons could be customised and moved around then you’ll have something great

That might seem a good idea at first but I cannot imagine how we could make that reliable with all those different platforms, wxWidgets versions, HiDPI screens and so on. Not to mention this would make support difficult as they would never know what the user sees. Distinguishing between Simple/Advanced/Expert modes is probably the furthest we dare to go.

guestisp, 2019-08-02:
Agree with @lukasmatena and @bubnikv
I like the current disposition