prusa3d/PrusaSlicer #2607: More Keyboard short cuts Needed in S2 or Object Menu returned

prowlmedia, 2019-07-06:
Right clicking then navigating to Mirror > X is painful when you have a lot to do - You can only perform this per single object. 

The object menu from the old version is now gone.  On a Mac at least I could assign ANY keyboard short cut I wanted to any menu item... now the Mirror is hidden under right click (why??) this is impossible. 

Also adding icons to menu items makes custom shortcuts impossible

traviselijah, 2019-07-24:
Mirror on an axis is the same as rotating it. Select one, or all, objects on the plate and press R, then drag the correct rotate block to rotate the object on the axis you'd like

prowlmedia, 2019-07-24:
> Mirror on an axis is the same as rotating it. Select one, or all, objects on the plate and press R, then drag the correct rotate block to rotate the object on the axis you'd like

Um... no Mirroring is not the same as rotating.  

For example - take an STL with Text on it for example. Rotate it anyway you want. The text on the is still correct - now look at it in a mirror.  Text is flipped and Backward right? 

Another example - no amount or rotating is going to give you the Left hand side of the car...
https://vimeo.com/55107711

Also even if you could drag a scale to -100% to flip it (which you cannot) ... selecting > Click on Scale icon > manually drag handles is a really long winded way to flip. 

Also scale and position have no snap to 5 or 10% or whatever makes them useless for a lot of things.  Rotate does have various percentage snaps ( but again cannot mirror an object )

traviselijah, 2019-07-24:
You are incorrect. If you rotate it on the Y or X axis though, you are "flipping" it. Trust me, I JUST had to do this for a project I am working on since I wanted to print letters on the textured PEI sheet with the textured surface being the front of the letters, so I had to rotate them. Just try it and you will see what I am talking about. When you press R, you get 3 circular items that show up on the screen, one along each axis. The red or green one will get you what you're looking for.

prowlmedia, 2019-07-24:
That ONLY works if the object is completely symmetrical in some way right? 

If it s a complex part that doesn't work.  All you are doing is rotating the object to print on a different face.  Not physically flipping the vertices of the model which is what mirroring is ( or scaling -100% ) I am guessing your text had a flat top and a flat bottom right - so you are just rotating to print on the top. 
![Mirror]

(https://user-images.githubusercontent.com/20207904/61786674-19f47a00-ae06-11e9-8f3b-421c69b7fa5a.png)

Attached 2 STLs... 
1 normal
2 Mirrored on X-Axis before export to STL. 

Feel tree to try Rotation to make object - 1 look like 2  ;)  Won't work in any way. 

Then right click > Mirror along X-Axis

[STLs.zip](https://github.com/prusa3d/PrusaSlicer/files/3426164/STLs.zip)

prowlmedia, 2019-07-24:
Tumbleweeds... guess you had an epiphany :)

bubnikv, 2019-08-01:
The upcoming PrusaSlicer 2.1.0-alpha1 implements mirroring buttons at the side bar (as of now only with the local coordinates active).

prowlmedia, 2019-08-01:
Local is awesome. appreciate this.

bubnikv, 2019-08-09:
Please close this issue after testing the upcoming PrusaSlicer 2.1.0-alpha1 if it solves your issue. Thanks.

prowlmedia, 2019-11-03:
This is great!  Please can we have those as shortcuts keys?
Suggest 
shift+X
shift+y
Shift+z

rtyr, 2021-02-03:
Issue with the same request (from the same author) https://github.com/prusa3d/PrusaSlicer/issues/2938. Closing this one.

closed 2021-02-03 by rtyr