prusa3d/PrusaSlicer #2608: Feature request: add automatic application updates

AlexanderSidorenko, 2019-07-06:
### Version
PrusaSlicer-2.0.0

### Operating system type + version
Ubuntu 18.04.2 LTS

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3

### Behavior
Hi Prusa team! You are doing absolutely outstanding job improving Prusa Slicer! Thank you!
As a minor downside of that, it can be hard to manually keep Prusa Slicer up to date.

Please consider adding an automated way to update application.

https://github.com/AppImage/AppImageUpdate
seems like a viable option for AppImage flavor.
Packaging Prusa Slicer as a .deb package and setting up package repository seems like another viable option.

#### Project File (.3MF) where problem occurs
N/A

probonopd, 2019-09-21:
It would even be possible to incorporate a small GUI for libappimageupdate so that one could update PrusaSlicer from within its own GUI:

https://docs.appimage.org/packaging-guide/optional/updates.html#by-using-libappimageupdate

mrjohnc, 2019-12-25:
I'd really appreciate this functionality

closed 2026-09-29 by prusaslicer-bot[bot]