prusa3d/PrusaSlicer #2610: Mac trackpad gestures

myreauks, 2019-07-07:
### PrusaSlicer Version 2.0.0+
### macOS 10.14.5

### Feature Request
Would it be possible in the future versions to support the standard Mac trackpad gestures in the 3d viewport? I mean pinch to zoom and two-finger drag to rotate the view. If this is already supported, is there a menu setting to turn it on?

myreauks, 2019-07-29:
A further question on this, is there any way to pan the 3d view with only a trackpad in use?

bubnikv, 2019-07-29:
These gestures are currently not supported. They are OSX specific. We may implement these gestures one day.

myreauks, 2019-08-05:
Thanks,
just to be sure, is there a way to pan the 3d view without a mouse?

soswow, 2020-08-24:
the only thing is missing is orbit when object is in the way. Currently single click and drag will orbit just fine. But if mouse down happens on object - then object moves. When I place support points very zoomed in there is simply no empty space (non-object) that I can mouse down on, to orbit.

bubnikv, 2020-10-28:
> the only thing is missing is orbit when object is in the way. Currently single click and drag will orbit just fine. But if mouse down happens on object - then object moves. When I place support points very zoomed in there is simply no empty space (non-object) that I can mouse down on, to orbit.

The PrusaSlicer 2.3.0-alpha3 will allow you to hold the Control key to rotate and pan the camera when inside the support point editing mode.

soswow, 2020-10-28:
> The PrusaSlicer 2.3.0-alpha3 will allow you to hold the Control key to rotate and pan the camera when inside the support point editing mode.

I just tried. I am not sure if I was clear with my initial message. Holding Control and dragging with normal click on trackpad pans. That wasn't a problem, since dragging with two fingers was already a panning action (no Control needed in this case). But rotation only possible when place you click when starting to drag is empty space and not a model.

soswow, 2020-10-28:
Ignore me. I have alpha0. Let me get the latest.

rufo, 2020-11-09:
I saw the release notes for 2.3.0 included updating to wxWidgets 3.1.3, and there's built-in [cross-platform support for multitouch gestures](https://docs.wxwidgets.org/trunk/classwx_gesture_event.html) now. Apologies if this was already known, just wanted to note it in case anyone decides to pick this up 🙂

bubnikv, 2020-11-10:
@rufo 
> I saw the release notes for 2.3.0 included updating to wxWidgets 3.1.3, and there's built-in cross-platform support for multitouch gestures now.

Thanks, we were not aware of that.
We will look into that.

myreauks, 2021-04-28:
> @rufo
> 
> > I saw the release notes for 2.3.0 included updating to wxWidgets 3.1.3, and there's built-in cross-platform support for multitouch gestures now.
> 
> Thanks, we were not aware of that.
> We will look into that.

Do you have plans on adding support for this in the future? It indeed seems that wxWidgets 3.1.3 onwards supports zoom gesture amonst others.

closed 2026-09-05 by myreauks