prusa3d/PrusaSlicer #2612: Multi material slicing issue

jwillmer, 2019-07-09:
### Version
2.0.0+win64

### Behavior
I try to slice a tower that uses two different extruders. I used Meshmixer to create (perfect) horizontal parts of my model.

I identified two problems:
- The wipe tower displays a print line in the air. Display bug?
- Slicer expects to use 7 tool changes. How does it get calculated? (Can I see the tool chain steps in slicer?) Is this a bug or can someone explain how to get to 7 changes?

[image]

#### Project File (.3MF) where problem occurs
[tower.zip](https://github.com/prusa3d/PrusaSlicer/files/3374547/tower.zip)

lukasmatena, 2019-07-10:
I think that everything is fine.

As for the first issue, you can look at #2412, #1355, #2088, #1616, #1070, #1268, #1423, #1386, #663 and #1581 (or the [FAQ](https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)) wiki page).

Figuring out the number of tool changes is also easy - go layer by layer through the preview and count the tool changes. There are seven of them, which is why the Sliced Info shows seven. The reason there are not just three (as you probably expect) is that your "perfectly horizontal" parts are actually horribly wavy and the volumes partially intersect. So there are several layers where PrusaSlicer slices more than one volume and since the volumes differ in extruder, extra toolchanges are forced. All this is clearly seen in the preview.

It might be easier to load the tower as one object (I suppose that's how it started) and enclose the red parts in a slab modifier mesh and assign different extruder to it. That way you won't have to worry about the cuts at all.

If your needs are satisfied, please, close the issue.

jwillmer, 2019-07-10:
Cool, `slab` is the missing piece. I will play around with this. Thank you!

closed 2019-07-10 by jwillmer

lukasmatena, 2019-07-11:
As pointed out by @YuSanka, in this simple case you could also consider using single extruder profile, mark the layers with colorprint feature and then do the three changes manually. That way you would get rid of the wipe tower altogether and save half of the needed filament.

jwillmer, 2019-07-11:
@lukasmatena thx for the tip! I will give it a try :)

jwillmer, 2019-07-11:
@lukasmatena I noticed that the [color-print-tool](https://blog.prusaprinters.org/color-print/) does not recognize the new Prusa 2.0.0 slicer. Do u think this is a problem or shall I just give it a try?

lukasmatena, 2019-07-11:
This is not a problem. That tool is obsolete so don't give it a try. Instead just read [this](https://blog.prusaprinters.org/prusaslicer-2-release/#10).