prusa3d/PrusaSlicer #2613: Sliced view looks good but prints extra lines

john-petrangelo, 2019-07-10:
### Version
PrusaSlicer-2.0.0+-201905201652

### Operating system type + version
macOS Mojave 10.14.5

### 3D printer brand / version + firmware version (if known)
Prusa MK3S, version 3.7.2
MMU2S, version 1.0.5

### Behavior
I'm trying to print the parts for the Lack table enclosure for MMU2S:
https://blog.prusaprinters.org/mmu2s-printer-enclosure/

Trying to print the first parts file:
https://media.prusaprinters.org/media/prints/3673/stls/23622_6c89f0b1-354c-414a-8e2a-7d7a8b45fd72/prusa_mmu_enclosure_1.3mf

I loaded the file into PrusaSlicer. Setup for MMU2S with single material, PLA. My spool was nearly depleted, so I deleted some of the parts to have a smaller print, leaving only the right rear corner and two hinges. I added a brim because the hinges have a small contact area. I clicked the Arrange button, sliced, generated G-code, and started printing. It looked ok at first, but after a few minutes I could see it was printing some extra lines of plastic that didn't belong there and were not in the sliced view in PrusaSlicer viewer. I'm pretty sure this is a GCode generation bug.

### Files

[Enclosure-BadGCode.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3375324/Enclosure-BadGCode.3mf.zip)

[Enclosure-BadGCode.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/3375337/Enclosure-BadGCode.gcode.zip)

[image]

john-petrangelo, 2019-07-10:
Tonight I tried the same procedure with an older build of the software I still have around, so far it looks better after several layers.
Slic3rPE-1.41.3+full-201902131031

lukasmatena, 2019-07-10:
I cannot tell for sure but to me it seems like a firmware issue. I didn't see these lines when I generated a preview of your gcode and even though preview softwares are often unreliable, if PrusaSlicer also showed correct preview, I'd guess the gcode is fine. I'll try to print your gcode here when a printer is available.

lukasmatena, 2019-07-12:
We printed your gcode and this is what we got (fw 3.7.2, mmu 1.0.5). As far as I can tell, there is nothing wrong with the gcode, so this is unlikely to be a PrusaSlicer issue and the cause must be somewhere else. I'll discuss it with the fw guys.
[image]

bubnikv, 2019-08-09:
Likely a firmware issue. Please contact the firmware team.

closed 2019-08-09 by bubnikv

DRracer, 2019-08-09:
FW 3.8 final will solve this