prusa3d/PrusaSlicer #2614: Smarter object ordering

haarp, 2019-07-10:
PrusaSlicer 2.0 on Gentoo Linux. This is a feature request.

PrusaSlicer tends to order the way in which it will print parts of an object, or multiple objects sub-optimally. Take a look at this screenshot (project: [PartA.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3378322/PartA.3mf.zip)):

[image]

PSlicer will print them in order: 1, 2, 3, 4, layer change, 1, 2, 3, 4, etc.

This leads to unnecessary time spent travelling and oozing.

A better approach would be: 1, 2, 4, 3, layer change, 3, 4, 2, 1, layer change, 1, 2, 4, 3, etc.

Preferring the nearest part after a layer change, instead of starting over at 1 every time.

I'm sure there's a reason why PSlicer is not doing this already. Maybe trying to avoid overheating the part? But to prevent that, there's already the slow-down feature in filament settings. If there's already an open issue for this, please point me in that way, I couldn't find it.

Thanks a lot!

DrLex0, 2019-07-17:
Always printing the parts in the same order produces a more consistent result. Otherwise there will be a local risk of overheating at positions 1 and 3 when the next layer is printed on top of the freshly extruded warm material, while positions 2 and 4 always get more time to cool down. Even if it does not overheat, it can cause the appearance of those parts to differ. The slow-down feature will do nothing about this because it only considers the entire layer.

For a ‘quick & dirty‘ print setting I would agree that 1243 and then 3421 would be best, but for best quality, the same ordering in every layer should be preserved. But then the ordering must indeed be 1243 instead of 1234…

haarp, 2019-07-21:
Closing this as dupe of #2601

closed 2019-07-21 by haarp