prusa3d/PrusaSlicer #2615: [Feature Request] Provide option to generate fan control (M106) as a percentage duty cycle instead of 0-255

mvadu, 2019-07-10:
I opened a [new issue for main line Slic3r](https://github.com/slic3r/Slic3r/issues/4818) to get a new option added to Printer setup screen where we can indicate the fan control signal uses hardware PWM which uses 0-100% duty cycle instead of software PWM which uses 0-255 as duty cycle. They are ready to add it to the printer advanced configuration and use it for `M106` generation. Since Prusa Slicer is on brand new UI layer, thought of opening the issue here as well.

**Is your feature request related to a problem? Please describe.**
I am using Smoothieware as my 3d printer platform. In my case I am specifically using a pin which generates hardware PWM, where the `S` value denotes the duty cycle in percent. In the Prusa Slicer cooling control I can set the max value for the fan control, but it always assumes the `S` value for `M106` to be in a 0-255 scale. 

**Describe the solution you'd like**
Having an option to define (May be at printer level as the flavor of M106 is more board specific i.e. whether board is using software or hardware PWM than the filament specific) will be great for this case.

**Describe alternatives you've considered**
I can manually calculate actual maximum duty cycle allowed (example for a specific filament I want to run the cooling only up to 89%) using max%/255*100 (89/255*100=~35%) and entering that in the filament config section as max percentage. 

**Additional context**
My current solution works, but that makes the filament configuration not portable. I can't share my filament config with anyone who is not using same setup of hardware PWM like mine. Before sharing I have to remember to update the config to 89% in my example.

mvadu, 2019-07-10:
https://github.com/slic3r/Slic3r/tree/feature-4818 shows how it is done on Slic3r

bubnikv, 2020-11-30:
Does smoothieware always accept fan speed in percent instead of (0,255) interval?
If so, then we should just emit the fan speed in percent for smoothieware only by default.

mvadu, 2020-12-01:
@bubnikv no, Smoothieware translates the M106 to a PWM, and based on the pin used it can be a software PWM or a hardware PWM. In case of software PWM it requires usual 0-255 range, and if its a hardware PWM it expects it in percentage.
Hence the ask for having a configurable output type for fan speed.

Ref: [switch.module_name.output_type](http://smoothieware.org/switch)

bubnikv, 2020-12-01:
>  Smoothieware translates the M106 to a PWM, and based on the pin used it can be a software PWM or a hardware PWM. In case of software PWM it requires usual 0-255 range, and if its a hardware PWM it expects it in percentage.

That is quite a mess. So it should be a checkbox then, and only provided for Smoothieware.

Th3M1k3y, 2023-07-08:
I found this old feature request by searching before creating a new one.

I would like having an option to change it between 0.0-1.0 and 0-255 as well. With newer brushless motors, they are mostly running far from at 100%, and many seems to have a rather large dead zone from zero until they start too. Adding the option for 0.0-1.0 scale and rounding off to 5 decimals, should close this once and for all, as the 0-1 easily could be converted into 0-65536, which should give plenty of resolution.

RepRapFirmware expect M106 S1 to be on the 0-1 scale, so that will turn the fan on at 100%, and M106 S2 will be on the 0-255 scale.

closed 2026-09-29 by prusaslicer-bot[bot]